agherzan/meta-raspberrypi #536: build vc-graphics failed in yocto project

limeng-linux, 2019-12-10:
I download meta-raspberrypi layer, and put it into poky, version is zues.
I want to use hardware accelerator based on GPU, so I build vc-graphics package with below comand

# bitbake vc-graphics

But some binary tool link very old lib, and cause error as below:

ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires libc.so.6(GLIBC_2.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires libpthread.so.0, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires libdl.so.2, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires librt.so.1, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires libc.so.6, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/dtoverlay-pre contained in package vc-graphics requires /bin/bash, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcsmem contained in package vc-graphics requires libpthread.so.0(GLIBC_2.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libgcc_s.so.1(GCC_3.5), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libgcc_s.so.1, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/raspiyuv contained in package vc-graphics requires libdl.so.2(GLIBC_2.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/raspistill contained in package vc-graphics requires libm.so.6(GLIBC_2.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/raspiyuv contained in package vc-graphics requires librt.so.1(GLIBC_2.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libm.so.6, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libstdc++.so.6(CXXABI_1.3), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libstdc++.so.6(GLIBCXX_3.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libgcc_s.so.1(GCC_3.4), but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: /usr/bin/vcdbg contained in package vc-graphics requires libstdc++.so.6, but no providers found in RDEPENDS_vc-graphics? [file-rdeps]
WARNING: vc-graphics-20190823-r1.0 do_package_qa: QA Issue: https://github.com/raspberrypi/firmware/archive/7163480fff007dc98978899b556dcf06f8a462c8.tar.gz      file://egl.pc     file://vchiq.sh : SRC_URI uses unstable GitHub archives [src-uri-bad]
ERROR: vc-graphics-20190823-r1.0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: /buildarea2/wr1019-build-test/rpi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/vc-graphics/20190823-r1.0/temp/log.do_package_qa.16567
ERROR: Task (/buildarea2/wr1019-build-test/poky-zeus/meta-raspberrypi/recipes-graphics/vc-graphics/vc-graphics.bb:do_package_qa) failed with exit code '1'

could you please help to check this issue?
Do I use the correct package for hardware accelerator feature?

thanks,
Limeng

limeng-linux, 2019-12-10:
Hi Andrei,

Could you please help to check which is the real HW accelerator lib, vc-graphics or userland?
Whether raspberry pi 4 platform supports HW accelerator based on GPU.
If vc-graphics is the correct one.
could you please tell me how to build vc-graphics in ycoto project?

thanks,
Limeng

kraj, 2019-12-10:
what are you using for libc ?

limeng-linux, 2019-12-11:
glibc 2.30 in my system.

kraj, 2019-12-11:
I means `TCLIBC` variable in yocto. You can find that with `bitbake -e vc-graphics|grep -e "^TCLIBC=`

limeng-linux, 2019-12-11:
mli1@pek-lpggp4:/buildarea2/wr1019-build-test/rpi/build$ bitbake -e vc-graphics|grep -e "^TCLIBC="
TCLIBC="glibc"

kraj, 2019-12-11:
Add following to the recipe and see if that helps

```
INSANE_SKIP_${PN} += "file-rdeps"
```

limeng-linux, 2019-12-11:
yes! I changed ERROR_QA to WARNING_QA for this package. but when I build the whole image, there are other errors, one example as below:

NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'lib32-libgl' (but virtual:multilib:lib32:/buildarea2/wr1019-build-test/bcm-2xxx-rpi-wr1019-build/layers/oe-core/meta/recipes-gnome/gtk+/gtk+3_3.24.8.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'lib32-libgl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['lib32-libgl']
ERROR: Required build target 'lib32-gtk+3' has no buildable providers.
Missing or unbuildable dependency chain was: ['lib32-gtk+3', 'lib32-libgl']

because I set libgles2 and egl to "vc-graphics", there is above error.
PREFERRED_PROVIDER_virtual/libgles2 = "vc-graphics"
PREFERRED_PROVIDER_virtual/egl = "vc-graphics"

I am not sure whether "vc-graphics" is verified with yocto project or how to use "vc-graphics" package

kraj, 2019-12-11:
I think you should disable multilib. Or use `userland`

limeng-linux, 2019-12-11:
could you please send me guide doc to show how to use userland or how to enable hardware accelerator feature?

thanks,
Limeng

agherzan, 2019-12-17:
@limeng-linux any reason to not use vc4?

limeng-linux, 2019-12-18:
Yes!
I have used, it works fine now.

Thanks,
Limeng

pbrkr, 2020-01-16:
Closing this based on the comment above from @limeng-linux. Feel free to re-open if this isn't resolved.

closed 2020-01-16 by pbrkr