agherzan/meta-raspberrypi #538: Cannot enable eglfs on Raspberry Pi `qt.qpa.plugin: Could not find the Qt platform plugin "eglfs" in ""`

adrianalin, 2019-12-14:
I am trying to build a qt5 basic image (with Qt5.13.2) for Raspberry Pi3 using the latest master of `meta-qt5`. When I try to run my python application (which uses python3-pytqt5, and Qml), I get:
```
root@raspberrypi3:~/hadesk# python3 main.py
qt.qpa.plugin: Could not find the Qt platform plugin "eglfs" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: minimal, offscreen, vnc, xcb.

Aborted

```
If I look at `build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/qtbase/5.13.2+gitAUTOINC+a7a24784ee-r0/build/config.summary` looks like EGLFS backend for Raspberry Pi3 is not enabled:
```
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. no
  LinuxFB ................................ no
  VNC .................................... yes
  Mir client ............................. no
  XCB:
    Using system-provided XCB libraries .. yes
    XCB XKB .............................. yes
    XCB XInput ........................... yes
    Native painting (experimental) ....... yes
    GL integrations:
      GLX Plugin ......................... yes
        XCB GLX .......................... yes
      EGL-X11 Plugin ..................... yes
```
In my own layer I added `qtbase_%.bbappend` with the following content:
```
PACKAGECONFIG_append = " accessibility eglfs fontconfig gles2 linuxfb tslib"

PACKAGECONFIG_remove = "examples tests"
DEPENDS += "userland"
```
In my `build/conf/local.conf` I also added
```
MACHINE = "raspberrypi3"
DISTRO_FEATURES_remove = " X11 wayland"
DISTRO_FEATURES_append = " wifi eglfs opengl usbhost ${DISTRO_FEATURES_LIBC}"
MCHINE_FEATURES_remove = "apm"
DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"

DISTRO ?= "poky"

PACKAGE_CLASSES = "package_ipk"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

USER_CLASSES ?= "buildstats image-mklibs image-prelink"

PATCHRESOLVE = "noop"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"
```

kraj, 2019-12-14:
this is not rpi isssue but distro problem. Look into Yoe e.g. where such issues are handled already

https://github.com/YoeDistro/yoe-distro

closed 2019-12-16 by adrianalin

adrianalin, 2019-12-16:
Found the `meta-raspberrypi/dynamic-layers/qt5-layer/recipes-qt/qt5/qtbase_%.bbappend` where are the variables needed to set/unset.
I modified my `build.conf/local.conf`:
```
DISTRO_FEATURES_remove = "x11"
MACHINE_FEATURES_remove = "vc4graphics"
```