agherzan/meta-raspberrypi #539: raspberrypi4-64 kernel

miapisquare, 2019-12-17:
**Raspberrypi4-64 kernel compile failure**

**Steps to reproduce the issue:**
***local.conf***
```
MACHINE = "raspberrypi4-64"

DL_DIR = "${HOME}/ssd/oe-sources"
SSTATE_DIR = "${HOME}/ssd/sstate-cache"
TMPDIR = "${HOME}/ssd/tmp-warrior"

GPU_MEM = "256"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_FSTYPES = "rpi-sdimg"
IMAGE_INSTALL_append = " packagegroup-core-x11-base mesa-demos kmscube cinematicexperience
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " x11"
DISTRO_FEATURES_remove = " wayland"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"

DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
USER_CLASSES ?= "image-mklibs image-prelink"
DISABLE_OVERSCAN = "1"
INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P secret root; "

```
***bblayers.conf***
```POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
    ${HOME}/poky-warrior/meta \
    ${HOME}/poky-warrior/meta-poky \
    ${HOME}/poky-warrior/meta-openembedded/meta-xfce \
    ${HOME}/poky-warrior/meta-openembedded/meta-gnome \
    ${HOME}/poky-warrior/meta-openembedded/meta-oe \
    ${HOME}/poky-warrior/meta-openembedded/meta-multimedia \
    ${HOME}/poky-warrior/meta-openembedded/meta-networking \
    ${HOME}/poky-warrior/meta-openembedded/meta-perl \
    ${HOME}/poky-warrior/meta-openembedded/meta-python \
    ${HOME}/poky-warrior/meta-qt5 \
    ${HOME}/poky-warrior/meta-raspberrypi \
    ${HOME}/rpi/meta-rpi \
    ${HOME}/rpi/meta-web-kiosk \
    ${HOME}/rpi/meta-webkit \
"
```

```
bitbake core-image-base
```

```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
[... 62 lines cut ...]
Summary: 1 task failed:
  /home/ubuntu/poky-warrior/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.19.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

**Additional details (revisions used, host distro, etc.):**
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.3 LTS
Release:        18.04
Codename:       bionic
```

The build failed while compiling the linux kernel.
Should the device tree blob for bcm2708-rpi-zero really be compiled when compiling a kernel for the raspberrypi4? Im wondering why.

To be honest, does someone know how to fix that?

Im trying to build a working GPU accelerated x11 image for weeks now (With help of  for example #476). Does someone have a working configuration to get it working? 32bit/64bit does not matter. If yes, why dont we add it to the repository as example image?

I really appreciate every hint i get.

With kind regards,
Mia Metzler

agherzan, 2019-12-17:
Yes. The build system compiles all the rpi dtbs. So that is expected. There was a change at some point where they renamed it from "0" to "zero". I'm wondering if that is the case here. have you tried to check it against zeus?

miapisquare, 2019-12-19:
Im sorry that i wasted your time! I diged into the deep core of yocto today.
I found out that some of my following layers must issue the problems id have:

- meta-rpi
- meta-web-kiosk
- meta-webkit

With a bare metal yocto setup the compile process worked absolutely fine.
I will dig deeper into it next week.  I will try to commit patches to the given repositorys when i found the problem. (I guess its meta-rpi because of its linux-raspberrypi_4.19.bbappend)

But for now i will close this issue.
Thank you very much for your help!

With kind regards,
Mia

closed 2019-12-19 by miapisquare