agherzan/meta-raspberrypi #547: brcmfmac fails on boot

timothyfroehlich, 2020-01-03:
**Description**
I was updating my yocto build to use the latest warrior releases from the base yocto, meta-oe, etc, and started getting the below failures on boot without any additional configuration after flashing the SD card. I had updated my meta-raspberrypi to commit 2f0fa9a087b9ded28797d3087ccfa8f48bf5dd13 ( the commit form PR #541 ) and immediately began seeing the failure. I reverted the checkout to commit 5cb49f4d76a311201bd36cf6538e23a59e3d8030 and the problem went away.

While in this state I am unable to use the wifi. iproute2 reports:
`3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:f5:ff:50 brd ff:ff:ff:ff:ff:ff`

**Describe the results you received:**
`root@raspberrypi3:~# journalctl | grep brcm | head -n 30
Jan 01 00:00:06 localhost kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Jan 01 00:00:06 localhost kernel: brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
Jan 01 00:00:06 localhost kernel: usbcore: registered new interface driver brcmfmac
Jan 01 00:00:09 localhost kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May  2 2019 02:46:17 version 7.45.189 (r714228 CY) FWID 01-e1db26e2
Jan 01 00:00:09 localhost kernel: brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.136 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2019-05-02 02:31:27 
Jan 03 19:41:25 raspberrypi3 NetworkManager[241]: <info>  [1578080485.7238] rfkill0: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0) (driver brcmfmac)
Jan 03 19:41:29 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:29 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Jan 03 19:41:31 raspberrypi3 btuart[187]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Jan 03 19:41:31 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:36 raspberrypi3 kernel: brcmfmac: brcmf_dongle_scantime: Scan assoc time error (-110)
Jan 03 19:41:39 raspberrypi3 kernel: brcmfmac: brcmf_netdev_open: failed to bring up cfg80211
Jan 03 19:41:42 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:42 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Jan 03 19:41:44 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:44 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Jan 03 19:41:47 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:47 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Jan 03 19:41:49 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:49 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Jan 03 19:41:52 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:52 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Jan 03 19:41:55 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:55 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Jan 03 19:41:57 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:41:57 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Jan 03 19:42:00 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:42:00 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Jan 03 19:42:02 raspberrypi3 kernel: brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jan 03 19:42:02 raspberrypi3 kernel: brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
`

**Additional details (revisions used, host distro, etc.):**
Raspberry Pi 3B+, no USB or ethernet devices attached, serial port attached (ENABLE_UART=1), power from a proper RPi power adapter.
Repos used:
git://git.yoctoproject.org/poky: d0f73121551dc98f6924cd77952bf9ebf5ef3dd7
git://git.openembedded.org/meta-openembedded: a24acf94d48d635eca668ea34598c6e5c857e3f8
git://git.yoctoproject.org/meta-raspberrypi: 2f0fa9a087b9ded28797d3087ccfa8f48bf5dd13
https://github.com/mendersoftware/meta-mender.git: 9650b37762443f01cf694deb1e00417751a2a719

Kernel used: 4.14.112 (Mender updates don't play nice with the firmware being on a separate, unupdatable boot partition and changing formats so I'm using the kernel I used with Yocto 2.6)
RootFS is read only

Possibly relevant Yocto config:
`PREFERRED_VERSION_linux-raspberrypi ??= "4.14.%"
RPI_KERNEL_DEVICETREE = "\
    bcm2710-rpi-3-b.dtb \
    bcm2710-rpi-3-b-plus.dtb \
"
DISTRO_FEATURES += "bluez5 bluetooth wifi"
require conf/distro/include/yocto-uninative.inc
INHERIT += "uninative"

agherzan, 2020-01-13:
Are you able to reproduce this issue on `core-image-base`?

agherzan, 2020-07-06:
Closing due to lack of feedback.

closed 2020-07-06 by agherzan