agherzan/meta-raspberrypi #552: No tvservice on Raspberrypi 3 on Warrior Branch

ikkysleepy, 2020-01-11:
The following config and layers work fine for "Thud" branch but when switching to "Warrior" branch I noticed two things:

1. tvservice was not available
2. cec-client does not work and says

> No device type given. Using 'recording device'
> CEC Parser created - libCEC version 4.0.4
> no serial port given. trying autodetect: FAILED

There are no errors in the build script either. 

```
# Mender Specific
INHERIT += "mender-full"
MENDER_ARTIFACT_NAME = "release-2020.01.03"
MACHINE ??= "raspberrypi3"
MENDER_BOOT_PART_SIZE_MB = "40"
MENDER_PARTITION_ALIGNMENT = "4194304"

# Extra Storage for php-int
MENDER_STORAGE_TOTAL_SIZE_MB = "1724"
IMAGE_ROOTFS_MAXSIZE = "2500000"

# RPI Specific
RPI_USE_U_BOOT = "1"
#IMAGE_INSTALL_append = " kernel-image kernel-devicetree"
DISABLE_RPI_BOOT_LOGO = "1"
IMAGE_FSTYPES_remove += " rpi-sdimg"
DISTRO_FEATURES_append = " systemd wifi bluez5 bluetooth"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
ENABLE_UART = "1"

# App Specific
IMAGE_INSTALL_append = " kernel-image kernel-devicetree kernel-modules linux-firmware-rpidistro-bcm43455 linux-firmware-rpidistro-bcm43430 wpa-supplicant iw bash hostapd dnsmasq iptables avahi-daemon fcgi php php-cli php-fpm nginx nmap android-tools python3 nodejs icu wol libcec initscript  ntp nodejs-npm util-linux bluez5 openssh mosquitto inotify-tools"

# General Info
DISTRO ?= "poky"
CONF_VERSION = "1"
DL_DIR ?= "/portable/yocto/rocko/downloads"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
PACKAGE_CLASSES ?= "package_rpm"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

# In order to build from the repository that Jenkins has cloned, instead of
# letting bitbake pull the sources, we use "externalsrc".
INHERIT += "externalsrc"

# Host Name
hostname_pn-base-files = "Eddy"
```

# BBlayer
```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /portable/yocto/rocko/poky/meta \
  /portable/yocto/rocko/poky/meta-poky \
  /portable/yocto/rocko/poky/meta-yocto-bsp \
  /portable/yocto/rocko/meta-openembedded/meta-oe \
  /portable/yocto/rocko/meta-openembedded/meta-perl \
  /portable/yocto/rocko/meta-openembedded/meta-python \
  /portable/yocto/rocko/meta-openembedded/meta-networking \
  /portable/yocto/rocko/meta-openembedded/meta-webserver \
  /portable/yocto/rocko/poky/meta-mender/meta-mender-core \
  /portable/yocto/rocko/meta-raspberrypi \
  /portable/yocto/rocko/poky/meta-mender/meta-mender-raspberrypi \
  /portable/yocto/rocko/meta-security \
  "
```
**Steps to reproduce the issue:**
1. Create Yocto image using bblayer / local config
2. burn image to sd card
3. test on pi that tvservice is not available

So the HDMI output from the pi does work, but looks like it might be a bug in the hdmi / linux firmware process.

agherzan, 2020-01-13:
This is most probably due to vc4 default switch: https://github.com/agherzan/meta-raspberrypi/commit/690bdca57422447e49d4ef43862bf675e9acc28f#diff-c02a029d89d92b2ddeb5e523a8cd535b

ikkysleepy, 2020-01-14:
Thanks for the help. The warrior branch already has the `DISABLE_VC4GRAPHICS ` fix, so I made the variable equal to 1 in the local config file. I rebuilt the image and both tvservice and HDMI work via libcec;however, everything is now yellow on the screen. The bootup is fine but after the screen turns yellow and you can't see anything. Doing a tvservice -p makes the screen back to black but then the input section is gone. Something is not right.

ikkysleepy, 2020-01-14:
So the issue with the yellow screen was u-boot related.  To fix it you need to make your own patch by removing some lines, as seen here: https://github.com/mendersoftware/uboot-mender/commit/50f05ab10e9590f22cd07382821a7a7721825b36

My issue was fixed for patching the u-boot layer for meta-mender layer. See here for more info:
https://hub.mender.io/t/yellow-screen-on-meta-mender-using-warrior-branch-with-disable-vc4graphics-u-boot-patch/1438

closed 2020-01-14 by ikkysleepy