agherzan/meta-raspberrypi #556: RPI_KERNEL_DEVICETREE depending of the machine ?

myfreescalewebpage, 2020-01-13:
**Description**

Currently all device tree are compiled. It should be cleaner with correct device tree compiled depending of the machine selected.

My proposition is to make "RPI_KERNEL_DEVICETREE" machine dependent. If the feature is interesting I can submit a pull request on the branch you want.

**Steps to reproduce the issue:**

This is not an issue, just a proposition.

**Describe the results you received:**

All device trees are currently built.

**Describe the results you expected:**

Only the expected device tree should be produced.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

agherzan, 2020-01-22:
I don't see an issue in doing that - I also don't see a great benefit in doing so.

kraj, 2020-01-22:
I would also suggest to look if this makes sense as a core feature and should be in OE-Core rather

myfreescalewebpage, 2020-01-23:
Not a important benefit, just cleaning and preserve few hundred of bytes in the FAT partition for doing something else. Anyway actually I have my own machine in which I have defined RPI_KERNEL_DEVICETREE and RPI_KERNEL_DEVICETREE_OVERLAYS for my target, that's also doing the job.

agherzan, 2020-01-23:
So you've done this for purely for space on the boot partition, right?

myfreescalewebpage, 2020-01-23:
To have something clean. There is more and more variants of the raspberry pi, and it will soon no more be acceptable to build and install all the device trees just for the pleasure.

agherzan, 2020-01-24:
I was referring to RPI_KERNEL_DEVICETREE (as per the issue raise). Those should be a handful of them that's why I was pointing at space as it doesn't seem to be a real gain. Nevertheless, I'm happy to accept this split on a per-machine basis (as a PR).

agherzan, 2021-10-30:
Closing as no activity happened for a while.

closed 2021-10-30 by agherzan