agherzan/meta-raspberrypi #557: rpi-basic-image.bb: Unable to determine endianness for architecture 'INVALID'

fstrati, 2020-01-14:
Hello,

I'm trying to build core-image-minimal for raspberrypi4-64 on the zeus branch
and I had to create the file sanity.conf and I'm getting this errors:

fstrati$ bitbake -c cleanall core-image-minimal
Loading cache: 100% |##########################################################################################################| Time: 0:00:00
Loaded 28 entries from dependency cache.
ERROR: rpi-basic-image.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: rpi-basic-image.bb: Please add your architecture to siteinfo.bbclass
ERROR: rpi-hwup-image.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: rpi-hwup-image.bb: Please add your architecture to siteinfo.bbclass
ERROR: Failed to parse recipe: rpi-hwup-image.bb
ERROR: vte_0.56.3.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: vte_0.56.3.bb: Please add your architecture to siteinfo.bbclass

I don't understand why core-image-minimal is using such deprecated recipes.

do I need to switch to the master branch?

Forgive me if this is a dumb question...

kraj, 2020-01-14:
what is your build configuration that bitbake prints at the top when beginning a build ?

fstrati, 2020-01-14:
in local.conf:
BB_NUMBER_THREADS = "2"
# Also, make can be passed flags so it run parallel threads e.g.:
PARALLEL_MAKE = "-j 2"
MACHINE ?= "raspberrypi4-64.conf"

fstrati@fstrati-HP-Laptop-15-db0xxx:/media/fstrati/Bkp20191226/yocto_try/poky$ inxi -F
System:    Host: fstrati-HP-Laptop-15-db0xxx Kernel: 5.0.0-37-generic x86_64 bits: 64 Desktop: Gnome 3.28.4
           Distro: Ubuntu 18.04.3 LTS
Machine:   Device: laptop System: HP product: HP Laptop 15-db0xxx v: Type1ProductConfigId serial: N/A
           Mobo: HP model: 84AC v: 85.20 serial: N/A UEFI: Insyde v: F.11 date: 08/20/2018
Battery    BAT1: charge: 41.4 Wh 100.0% condition: 41.4/41.0 Wh (101%)
CPU:       Dual core AMD A9-9425 RADEON R5 5 COMPUTE CORES 2C+3G (-MCP-) cache: 2048 KB
           clock speeds: max: 3100 MHz 1: 1824 MHz 2: 2299 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] Stoney [Radeon R2/R3/R4/R5 Graphics]
           Display Server: x11 (X.Org 1.20.4 ) driver: amdgpu Resolution: 1366x768@60.00hz
           OpenGL: renderer: AMD STONEY (DRM 3.27.0, 5.0.0-37-generic, LLVM 8.0.0) version: 4.5 Mesa 19.0.8
Audio:     Card-1 Advanced Micro Devices [AMD] Family 15h (Models 60h-6fh) Audio Controller
           driver: snd_hda_intel
           Card-2 Advanced Micro Devices [AMD/ATI] Device 15b3 driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k5.0.0-37-generic
Network:   Card-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller driver: r8169
           IF: enp2s0 state: up speed: 100 Mbps duplex: full mac: e4:e7:49:0c:f2:c8
           Card-2: Realtek RTL8821CE 802.11ac PCIe Wireless Network Adapter driver: rtl8821ce
           IF: wlo1 state: up mac: 28:3a:4d:07:8d:f1
Drives:    HDD Total Size: 1500.3GB (7.6% used)
           ID-1: /dev/sda model: CT500MX500SSD1 size: 500.1GB
           ID-2: USB /dev/sdb model: MQ04ABF100 size: 1000.2GB
Partition: ID-1: / size: 457G used: 69G (16%) fs: ext4 dev: /dev/dm-0
           ID-2: swap-1 size: 1.03GB used: 0.11GB (10%) fs: swap dev: /dev/dm-1
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 34.4C mobo: N/A gpu: 34.0
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 355 Uptime: 16:16 Memory: 4244.2/7500.6MB Client: Shell (bash) inxi: 2.3.56 

bitbake exits with error and gives no hints:

fstrati@fstrati-HP-Laptop-15-db0xxx:/media/fstrati/Bkp20191226/yocto_try/poky$ bitbake -e core-image-minimal
Loading cache: 100% |##########################################################################################################| Time: 0:00:00
Loaded 28 entries from dependency cache.
ERROR: /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb: Please add your architecture to siteinfo.bbclass
ERROR: Failed to parse recipe: /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb
ERROR: /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi/recipes-core/images/rpi-hwup-image.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi/recipes-core/images/rpi-hwup-image.bb: Please add your architecture to siteinfo.bbclass

Summary: There were 5 ERROR messages shown, returning a non-zero exit code.

texierp, 2020-01-14:
Hi @fstrati 

It seems your `MACHINE` configuration is wrong.

Have you tried with `MACHINE ?= "raspberrypi4-64"` instead ?

Pierre-Jean

fstrati, 2020-01-14:
You're right!
Thanks

fstrati, 2020-01-14:
But now I have got some other errors for gstreamer:

fstrati@fstrati-HP-Laptop-15-db0xxx:/media/fstrati/Bkp20191226/yocto_try/poky/build$ bitbake -c cleanall core-image-minimal
Loading cache: 100% |##########################################################################################################| Time: 0:00:00
Loaded 1328 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'libmms' (but /media/fstrati/Bkp20191226/yocto_try/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.16.1.bb DEPENDS on or otherwise requires it). Close matches:
  libsm
  libfm
  libcomps
NOTE: Runtime target 'gst-examples' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gst-examples', 'gstreamer1.0-plugins-bad', 'libmms']
NOTE: Runtime target 'packagegroup-core-tools-testapps' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-tools-testapps', 'gst-examples', 'gstreamer1.0-plugins-bad', 'libmms']
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'packagegroup-core-tools-testapps', 'gst-examples', 'gstreamer1.0-plugins-bad', 'libmms']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

and

fstrati@fstrati-HP-Laptop-15-db0xxx:/media/fstrati/Bkp20191226/yocto_try/poky/build$ bitbake -c cleanall rpilinux-image
Loading cache: 100% |##########################################################################################################| Time: 0:00:00
Loaded 30 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################| Time: 0:03:12
Parsing of 802 .bb files complete (30 cached, 772 parsed). 1328 targets, 64 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'faad2' (but /media/fstrati/Bkp20191226/yocto_try/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.16.1.bb DEPENDS on or otherwise requires it)
NOTE: Runtime target 'gst-examples' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']
NOTE: Runtime target 'packagegroup-core-tools-testapps' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-tools-testapps', 'gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']
ERROR: Required build target 'rpilinux-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['rpilinux-image', 'packagegroup-core-tools-testapps', 'gst-examples', 'gstreamer1.0-plugins-bad', 'faad2']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

texierp, 2020-01-14:
@fstrati, can you share the content of your `bblayers.conf` ?

fstrati, 2020-01-14:
```
fstrati@fstrati-HP-Laptop-15-db0xxx:/media/fstrati/Bkp20191226/yocto_try/poky/build$ cat conf/bblayers.conf 
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /media/fstrati/Bkp20191226/yocto_try/poky/meta \
  /media/fstrati/Bkp20191226/yocto_try/poky/meta-poky \
  /media/fstrati/Bkp20191226/yocto_try/poky/meta-yocto-bsp \
  /media/fstrati/Bkp20191226/yocto_try/poky/meta-raspberrypi \
  /media/fstrati/Bkp20191226/yocto_try/poky/meta-rpilinux \
  "

```
meta-rpilinux is for my image

but in local.conf I have also:

`EXTRA_IMAGE_FEATURES ?= "debug-tweaks dev-pkgs tools-sdk tools-debug eclipse-debug tools-profile tools-testapps"`

texierp, 2020-01-14:
Thanks !

It seems you need `meta-oe` in bblayers.conf.

See https://github.com/openembedded/meta-openembedded

kraj, 2020-01-14:
dependencies are documented here https://github.com/agherzan/meta-raspberrypi/#dependencies

agherzan, 2020-01-22:
Closing this as it seems to have been a configuration issue.

closed 2020-01-22 by agherzan