agherzan/meta-raspberrypi #559: Using gpio-poweroff causes hang the shutdown process.

james-bay, 2020-01-15:
**Description**
I built core-image-base image then I added [gpio-poweroff.dtbo](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/gpio-poweroff.dtbo) to the `overlays` directory on an sd card and added `dtoverlay=gpio-poweroff,gpiopin=20` at the end of `config.txt`. When I run `poweroff` on gpio pin 20 I get `active_high`, but shutting down hangs. I don't use any external hardware, but with an HDMI LCD, the result is the same: the screen does not go blank and on it, I see a frozen cursor.

**The result I received:**
I can observe the following, when I using serial communication:
```
...
[  OK  ] Stopped File System Check on Root Device.
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
[  OK  ] Started Power-Off.
[  OK  ] Reached target Power-Off.
[   17.594523] systemd-shutdow: 32 output lines suppressed due to ratelimiting
[   17.623221] systemd-shutdown[1]: Syncing filesystems and block devices.
[   17.639589] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   17.666205] systemd-journald[103]: Received SIGTERM from PID 1 (systemd-shutdow).
[   17.762828] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   17.782986] systemd-shutdown[1]: Unmounting file systems.
[   17.792201] [191]: Remounting '/' read-only in with options '(null)'.
[   17.811266] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   17.822952] systemd-shutdown[1]: All filesystems unmounted.
[   17.828629] systemd-shutdown[1]: Deactivating swaps.
[   17.834357] systemd-shutdown[1]: All swaps deactivated.
[   17.839674] systemd-shutdown[1]: Detaching loop devices.
[   17.850578] systemd-shutdown[1]: All loop devices detached.
[   17.856301] systemd-shutdown[1]: Detaching DM devices.
[   17.878325] reboot: Power down
[   21.081928] ------------[ cut here ]------------
[   21.086633] WARNING: CPU: 0 PID: 1 at drivers/power/reset/gpio-poweroff.c:47 gpio_poweroff_do_poweroff+0xd8/0xe0
[   21.096955] Modules linked in: brcmfmac brcmutil sha256_generic evdev bcm2835_v4l2(C) bcm2835_codec(C) joydev v4l2_mem2mem bcm2835_mmal_vchiq(C) hid_multitouch cfg80211 videobuf2_dma_contig v4l2_common videobuf2_vmalloc rfkill videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon videodev hwmon media vc_sm_cma(C) uio_pdrv_genirq uio fixed sch_fq_codel ipv6
[   21.130233] CPU: 0 PID: 1 Comm: systemd-shutdow Tainted: G         C        4.19.75 #1
[   21.138263] Hardware name: BCM2835
[   21.141734] [<80112158>] (unwind_backtrace) from [<8010d400>] (show_stack+0x20/0x24)
[   21.149599] [<8010d400>] (show_stack) from [<8081eef8>] (dump_stack+0xc8/0x10c)
[   21.157028] [<8081eef8>] (dump_stack) from [<80120d40>] (__warn.part.0+0xcc/0xe8)
[   21.164629] [<80120d40>] (__warn.part.0) from [<80120ef4>] (warn_slowpath_null+0x54/0x5c)
[   21.172934] [<80120ef4>] (warn_slowpath_null) from [<8067bdb0>] (gpio_poweroff_do_poweroff+0xd8/0xe0)
[   21.182298] [<8067bdb0>] (gpio_poweroff_do_poweroff) from [<8010ba9c>] (machine_power_off+0x38/0x3c)
[   21.191576] [<8010ba9c>] (machine_power_off) from [<80146124>] (kernel_power_off+0x7c/0x80)
[   21.200059] [<80146124>] (kernel_power_off) from [<801462e8>] (sys_reboot+0x168/0x1fc)
[   21.208101] [<801462e8>] (sys_reboot) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[   21.215870] Exception stack(0xb9d0dfa8 to 0xb9d0dff0)
[   21.220998] dfa0:                   4321fedc 00000000 fee1dead 28121969 4321fedc 0db59200
[   21.229301] dfc0: 4321fedc 00000000 00000000 00000058 00000000 00000003 00000000 00441604
[   21.237599] dfe0: 00000058 7e80ec94 76cf0785 76c75d16
[   21.242723] ---[ end trace 8facc6e9308951e0 ]---
```
**Additional details**
I use yocto branch Zeus, `systemd` as an init system and simple `local.conf`:
```
LICENSE_FLAGS_WHITELIST = "commercial"

DISTRO_FEATURES = "ext2 opengl usbhost ${DISTRO_FEATURES_LIBC}"

PREFERRED_PROVIDER_udev = "eudev"

# systemd
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"

MACHINE_FEATURES_remove = "apm"
MACHINE = "raspberrypi3"

KERNEL_IMAGETYPE = "zImage"

DISABLE_VC4GRAPHICS = "1"
DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
SERIAL_CONSOLES = "115200;ttyAMA0"

PREFERRED_VERSION_linux-raspberrypi = "4.19.%"

TMPDIR = "${HOME}/linux/yocto-zeus/rpi-minimal/build/tmp"
DL_DIR = "${HOME}/linux/yocto-zeus/download"
SSTATE_DIR = "${HOME}/linux/yocto-zeus/rpi-minimal/sstate-cache"

DISTRO = "poky"
PACKAGE_CLASSES = "package_ipk"

SDKMACHINE = "x86_64"

INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P root root; "

USER_CLASSES = "image-mklibs image-prelink"
PATCHRESOLVE = "noop"
RM_OLD_IMAGE = "1"
INHERIT += "rm_work"
CONF_VERSION = "1"

DISABLE_RPI_BOOT_LOGO = "1"
DISABLE_SPLASH = "1"
BOOT_DELAY = "0"
RPI_EXTRA_CONFIG ='\n\
# Poweroff detection \n\
dtoverlay=gpio-poweroff,gpiopin=20 \n\
'
```

agherzan, 2020-01-22:
@james-bay have you tried to replicate this on a Raspbian?

james-bay, 2020-01-24:
@agherzan No, but I will try. However, if I right understand [this](https://www.kernel.org/doc/Documentation/devicetree/bindings/power/reset/gpio-poweroff.txt):
```
After a delay of 100ms, the GPIO is set to inactive, thus causing an active->inactive edge, triggering negative edge triggered power off. After another 100ms delay the GPIO is driver active again. If the power is still on and the CPU still running after a 3000ms delay, a WARN_ON(1) is emitted.
```
It's expected behavior.

agherzan, 2020-01-24:
Got it! Thanks.

closed 2020-01-24 by agherzan