agherzan/meta-raspberrypi #567: "ENABLE_UART=0" do not disables console totally

myfreescalewebpage, 2020-01-20:
**Description**

When setting ENABLE_UART=0 to disable the console on GPIO14/15, the kernel boot log is not shown, but the console is started and the login prompt is displayed. It is possible to login and use the terminal.

**Steps to reproduce the issue:**
1. Build with ENABLE_UART=0
2. Load the raspberry pi (I'm using CM3+)
3. Open serial console and boot the pi
4. At the end of the boot the prompt is displayed on the console

**Describe the results you received:**

The console is available on the UART

**Describe the results you expected:**

The console should not be available at all on the UART, instead the UART should remain available for other usage

**Additional information you deem important (e.g. issue happens only occasionally):**

I have solved the issue setting "enable_uart=0" in the "config.txt" file. When ENABLE_UART=0 is set, the config file is just not filled. Instead it should be completed with "enable_uart=0".

**Additional details (revisions used, host distro, etc.):**

I'm using warrior branch

I can propose à PR if you indicate me on which branch I should do it. master ?

texierp, 2020-01-20:
Hi @myfreescalewebpage 

> I can propose à PR if you indicate me on which branch I should do it. master ? How to apply on all branches then ?

This must first be fixed on master and then backported to the specific branch with a new PR (with `git cherry-pick -x <commit>`)

Example:

- https://github.com/agherzan/meta-raspberrypi/pull/542
- https://github.com/agherzan/meta-raspberrypi/pull/543

Thanks !

myfreescalewebpage, 2020-01-20:
Hi @texierp thanks for the quick answer, will do that!

myfreescalewebpage, 2020-01-20:
https://github.com/agherzan/meta-raspberrypi/pull/568

lokeb, 2021-03-21:
Has this problem been fixed? I am observing the same problem on RPI-4 on Debian buster arm64.

closed 2021-11-14 by agherzan