agherzan/meta-raspberrypi #576: Wifi support for Pi 3 / Pi 0-W

pficheux, 2020-02-04:
Hi all,

What's the way to build a Wifi capable image  with Yocto 3.0 + meta-raspberrypi?
I got the following messages at boot time:

[    3.399597] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'       
[    3.407398] platform regulatory.0: Direct firmware load for regulatory.db fai
led with error -2                                                               
[    3.416231] cfg80211: failed to load regulatory.db                           
[    3.468006] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio f
or chip BCM43430/1                                                              
[    3.478060] usbcore: registered new interface driver brcmfmac                
[    3.507329] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430
-sdio.txt failed with error -2                                                  
[    3.509080] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)                    
Tue Feb  4 08:49:36 UTC 2020                                                    
[    3.869115] urandom_read: 2 callbacks suppressed                             
[    3.869124] random: dd: uninitialized urandom read (512 bytes read)          
INIT: Entering runlevel: 5                                                      
Configuring network interfaces... Successfully initialized wpa_supplicant       
rfkill: Cannot get wiphy information                                            
Could not read interface wlan0 flags: No such device                            
WEXT: Could not set interface 'wlan0' UP                                        
wlan0: Failed to initialize driver interface                                    
ip: SIOCGIFFLAGS: No such device                              
...

Added the following lines to conf/local.conf 

IMAGE_INSTALL_append = " iw linux-firmware-bcm43430 wpa-supplicant kernel-modules"

Then

$ bitbake core-image-minimal

I did the test on Pi 3 B/B+ and Pi 0 W.

Thx by advance

texierp, 2020-02-04:
Hi @pficheux 

It only appears with Zeus ?
Can you test with a Raspbian image ?

In any case, seems strange because I am on top-of master and it works fine on
my side:

````
root@raspberrypi3:~# ip -4 addr show wlan0
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel qlen 1000
    inet 192.168.1.51/24 brd 192.168.1.255 scope global wlan0
       valid_lft forever preferred_lft forever`
````

FYI, I use the following configuration with kas (`$: kas build kas-files/kas-poky.yml:kas-files/board-raspberrypi3.yml`)
- https://github.com/texierp/kas-files/blob/ci/kas-poky.yml
- https://github.com/texierp/kas-files/blob/ci/board-raspberrypi3.yml

Thanks

pficheux, 2020-02-04:
I see you use core-image-base. I've tested with core-image-minimal, may be not smart enough.

pficheux, 2020-02-04:
Works fine with core-image-base :-)

thks

closed 2020-02-04 by pficheux

texierp, 2020-02-04:
> Works fine with core-image-base :-)

Nice to hear that :+1: 

> thks

You're welcome ;)