agherzan/meta-raspberrypi #580: uImage-bcm2711-rpi-4-b.dtb No such file or directory

sirus20x6, 2020-02-13:
has the naming scheme changed with the pi 4? I have 
bcm2711-rpi-4-b-1-4.19.93+git0+3fdcc814c5-r0-raspberrypi4-20200213161019.dtb
bcm2711-rpi-4-b.dtb
bcm2711-rpi-4-b-raspberrypi4.dtb
but no uImage-bcm2711-rpi-4-b.dtb

`Initialising tasks: 100% |########################################################################################################################################################################################################################| Time: 0:00:08
Sstate summary: Wanted 23 Found 8 Missed 15 Current 2796 (34% match, 99% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: core-image-myImage-1.0-r0 do_image_wic: Execution of '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/tmp-glibc/work/raspberrypi4-oe-linux-gnueabi/core-image-myImage/1.0-r0/temp/run.do_image_wic.40339' failed with exit code 1:
INFO: Creating image(s)...

WARNING: bootloader config not specified, using defaults

ERROR: _exec_cmd: install -m 0644 -D /mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/tmp-glibc/deploy/images/raspberrypi4/uImage-bcm2711-rpi-4-b.dtb /mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/tmp-glibc/work/raspberrypi4-oe-linux-gnueabi/core-image-myImage/1.0-r0/deploy-core-image-myImage-image-complete/core-image-myImage-raspberrypi4-20200213161019/tmp.wic.35m3q06x/boot.1/bcm2711-rpi-4-b.dtb returned '1' instead of 0
output: install: cannot stat '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/tmp-glibc/deploy/images/raspberrypi4/uImage-bcm2711-rpi-4-b.dtb': No such file or directory

WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/tmp-glibc/work/raspberrypi4-oe-linux-gnueabi/core-image-myImage/1.0-r0/temp/log.do_image_wic.40339
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: 1943536.000000 = 1943536 * 1.000000
| DEBUG: 2045936.000000 = max(1943536.000000, 65536)[1943536.000000] + 102400
| DEBUG: 2045936.000000 = int(2045936.000000)
| DEBUG: 2048000 = aligned(2045936)
| DEBUG: returning 2048000
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/swig/swig_3.0.12.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/bison/bison_3.4.1.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-connectivity/openssl/openssl_1.1.1d.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/syslinux/syslinux_6.04-pre2.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/gcc/gcc-runtime_9.2.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/squashfs-tools/squashfs-tools_git.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/fdisk/gptfdisk_1.0.4.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-bsp/u-boot/u-boot-tools_2019.07.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/dnf/dnf_4.2.2.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/opkg/opkg_0.4.1.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/flex/flex_2.6.0.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-support/lzop/lzop_1.04.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-support/bmap-tools/bmap-tools_3.5.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/binutils/binutils-cross_2.32.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/icecc-create-env/icecc-create-env_0.1.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/btrfs-tools/btrfs-tools_5.2.2.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.1.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/python/python-native_2.7.17.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-extended/bc/bc_1.07.1.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/dosfstools/dosfstools_4.1.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-core/glibc/cross-localedef-native_2.30.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/mtools/mtools_4.0.23.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-raspberrypi/recipes-bsp/bootfiles/rpi-config_git.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/rpm/rpm_4.14.2.1.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-raspberrypi/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.19.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-core/glibc/glibc_2.30.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-myImage-system/recipes-kernel/dtc/dtc_1.4.5.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/gcc/gcc-cross_9.2.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-myImage-system/recipes-core/myImage-additional/myImage-additional_1.0.0.bb:do_populate_sysroot', '/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/bu [... cut ...]
ERROR: Task (/mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-myImage-system/recipes-core/images/core-image-myImage.bb:do_image_wic) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7633 tasks of which 7502 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /mnt/md0/myuser/svn/yocto/myImage-yocto-system/branches/mybranch/build/../poky/meta-myImage-system/recipes-core/images/core-image-myImage.bb:do_image_wic
Summary: There were 4 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

real	18m21.627s
user	0m4.616s
sys	0m0.532s
`

agherzan, 2020-02-20:
Can you please reformat your issue and make sure you clearly describe what your setup is, what you are building and how it breaks? The current report is undecipherable.

agherzan, 2020-07-06:
Closing due to lack of feedback.

closed 2020-07-06 by agherzan