agherzan/meta-raspberrypi #581: rpi-test-image fails to build

california68, 2020-02-13:
**Description**

When executing bitbake rpi-test-image for Raspberry Pi 1, build breaks with a failure

**Steps to reproduce the issue:**
1. Checkout poky, meta-openembedded and meta-raspberrypi on master branch
2. Adapt local.conf to machine "raspberrypi" and bblayers.conf to needed layers 
3. Execute bitbake rpi-test-image

**Describe the results you received:**
```
ERROR: rpi-test-image-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs --setopt=logdir=/home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/temp --repofrompath=oe-repo,/home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/oe-rootfs-repo --nogpgcheck install lighttpd packagegroup-base-extended packagegroup-core-boot packagegroup-rpi-test proftpd psplash-raspberrypi run-postinsts locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.2.2
cachedir: /home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/cw/Yocto/build/tmp/work/raspberrypi-poky-linux-gnueabi/rpi-test-image/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Thu 13 Feb 2020 06:34:01 PM UTC.
No module defaults found
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: package packagegroup-base-wifi-1.0-r83.raspberrypi requires wireless-regdb-static, but none of the providers can be installed
  - package wireless-regdb-2019.06.03-r0.noarch conflicts with wireless-regdb-static provided by wireless-regdb-static-2019.06.03-r0.noarch
  - package packagegroup-base-1.0-r83.raspberrypi requires packagegroup-base-wifi, but none of the providers can be installed
  - package packagegroup-rpi-test-1.0-r0.raspberrypi requires wireless-regdb, but none of the providers can be installed
  - package packagegroup-base-extended-1.0-r83.raspberrypi requires packagegroup-base, but none of the providers can be installed
  - conflicting requests
```

**Describe the results you expected:**
Successful build of rpi-test-image

**Additional information you deem important (e.g. issue happens only occasionally):**
Tested on two different build hosts and on "Zeus" and master branch with same issue

**Additional details (revisions used, host distro, etc.):**
Build host Ubuntu 18.04 LTS 64-bit

clem844, 2020-03-13:
Any update on this? I'm facing the same issue

texierp, 2020-03-13:
@clem844, @california68, can you try to replace `wireless-regdb` to `wireless-regdb-static` in 
`meta-raspberrypi/recipes-core/packagegroups/packagegroup-rpi-test.bb` to check if the build is fine with this modification ?

california68, 2020-03-16:
I replaced the package `wireless-regdb` with `wireless-regdb-static` and the build just works fine now

clem844, 2020-03-16:
I don't have the exact same problem as I don't build rpi-test-image but core-image-base. The error is quite the same though:

```

Could not invoke dnf. Command '/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs --setopt=logdir=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/temp --repofrompath=oe-repo,/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo --nogpgcheck install aktualizr aktualizr-auto-reboot aktualizr-log-debug aktualizr-shared-prov fit-conf kernel-modules libtss2 libtss2-tcti-device0 libtss2-tcti-mssim0 os-release ostree packagegroup-base-extended packagegroup-core-boot psplash-raspberrypi run-postinsts slb9670 systemd-journald-persistent tpm2-abrmd libtpm2-pkcs11-0 tpm2-tools virtual/network-configuration wifi-systemd-service locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.2.2
cachedir: /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Mon 16 Mar 2020 01:40:36 PM UTC.
No module defaults found
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: package packagegroup-base-extended-1.0-r83.raspberrypi3 requires packagegroup-base, but none of the providers can be installed
  - package packagegroup-base-1.0-r83.raspberrypi3 requires packagegroup-base-wifi, but none of the providers can be installed
  - conflicting requests
  - nothing provides wireless-regdb-static needed by packagegroup-base-wifi-1.0-r83.raspberrypi3
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

kraj, 2020-03-16:
@clem844 you might have to make same change in `meta/recipes-core/packagegroups/packagegroup-base.bb`

clem844, 2020-03-16:
It is already set to "wireless-regdb-static" in packagegroup-base.bb

kraj, 2020-03-16:
Do you have https://git.openembedded.org/openembedded-core/commit/?id=00c5a665b48c830a0008139b7ae1a51e79b72bb5 applied to oe-core ?

clem844, 2020-03-16:
yes, it is
```
git log | grep wireless-regdb
    wireless-regdb: Add recipe
    wireless-regdb-static should be used with kernel >= 4.15.
    wireless-regdb can be used with older kernels and is mostly
```

I'm on zeus branch

kraj, 2020-03-16:
dont see what could be wrong can you post `meta/recipes-core/packagegroups/packagegroup-base.bb`

clem844, 2020-03-17:
It's the same as master or zeus branch
**meta/recipe-core/packagegroups/packagegroup-base.bb**
```
SUMMARY = "Merge machine and distro options to create a basic machine task/package"
PR = "r83"

#
# packages which content depend on MACHINE_FEATURES need to be MACHINE_ARCH
#
PACKAGE_ARCH = "${MACHINE_ARCH}"

inherit packagegroup

PROVIDES = "${PACKAGES}"
PACKAGES = ' \
            packagegroup-base \
            packagegroup-base-extended \
            packagegroup-distro-base \
            packagegroup-machine-base \
            \
            ${@bb.utils.contains("MACHINE_FEATURES", "acpi", "packagegroup-base-acpi", "",d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "alsa", "packagegroup-base-alsa", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "apm", "packagegroup-base-apm", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "ext2", "packagegroup-base-ext2", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "vfat", "packagegroup-base-vfat", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "keyboard", "packagegroup-base-keyboard", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "pci", "packagegroup-base-pci", "",d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "pcmcia", "packagegroup-base-pcmcia", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "phone", "packagegroup-base-phone", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "serial", "packagegroup-base-serial", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "usbgadget", "packagegroup-base-usbgadget", "", d)} \
            ${@bb.utils.contains("MACHINE_FEATURES", "usbhost", "packagegroup-base-usbhost", "", d)} \
            \
            ${@bb.utils.contains("DISTRO_FEATURES", "bluetooth", "packagegroup-base-bluetooth", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "wifi", "packagegroup-base-wifi", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "3g", "packagegroup-base-3g", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "nfc", "packagegroup-base-nfc", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "cramfs", "packagegroup-base-cramfs", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "ipsec", "packagegroup-base-ipsec", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "ipv6", "packagegroup-base-ipv6", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "nfs", "packagegroup-base-nfs", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "ppp", "packagegroup-base-ppp", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "smbfs", "packagegroup-base-smbfs", "", d)} \
            ${@bb.utils.contains("DISTRO_FEATURES", "zeroconf", "packagegroup-base-zeroconf", "", d)} \
            \
            '

# Override by distro if needed
VIRTUAL-RUNTIME_keymaps ?= "keymaps"

#
# packagegroup-base contain stuff needed for base system (machine related)
#
RDEPENDS_packagegroup-base = "\
    packagegroup-distro-base \
    packagegroup-machine-base \
    \
    module-init-tools \
    ${@bb.utils.contains('MACHINE_FEATURES', 'apm', 'packagegroup-base-apm', '',d)} \
    ${@bb.utils.contains('MACHINE_FEATURES', 'acpi', 'packagegroup-base-acpi', '',d)} \
    ${@bb.utils.contains('MACHINE_FEATURES', 'keyboard', 'packagegroup-base-keyboard', '',d)} \
    ${@bb.utils.contains('MACHINE_FEATURES', 'phone', 'packagegroup-base-phone', '',d)} \
    \
    ${@bb.utils.contains('COMBINED_FEATURES', 'alsa', 'packagegroup-base-alsa', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'ext2', 'packagegroup-base-ext2', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'vfat', 'packagegroup-base-vfat', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'pci', 'packagegroup-base-pci', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'pcmcia', 'packagegroup-base-pcmcia', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'usbgadget', 'packagegroup-base-usbgadget', '',d)} \
    ${@bb.utils.contains('COMBINED_FEATURES', 'usbhost', 'packagegroup-base-usbhost', '',d)} \
[... 253 lines cut ...]

SUMMARY_packagegroup-base-zeroconf = "Zeroconf support"
RDEPENDS_packagegroup-base-zeroconf = "\
    avahi-daemon"
RDEPENDS_packagegroup-base-zeroconf_append_libc-glibc = "\
    libnss-mdns \
    "

SUMMARY_packagegroup-base-ipv6 = "IPv6 support"
RDEPENDS_packagegroup-base-ipv6 = "\
    "

RRECOMMENDS_packagegroup-base-ipv6 = "\
    kernel-module-ipv6 "

SUMMARY_packagegroup-base-serial = "Serial port support"
RDEPENDS_packagegroup-base-serial = "\
    setserial \
    lrzsz "

SUMMARY_packagegroup-base-phone = "Cellular telephony (voice) support"
RDEPENDS_packagegroup-base-phone = "\
    ofono"

```

texierp, 2020-03-17:
@clem844, are you able to share your setup ? (local.conf, bblayers.conf, ...)

clem844, 2020-03-17:
bblayer.conf
```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/tce/yocto_dist/poky/meta \
  /home/tce/yocto_dist/poky/meta-poky \
  /home/tce/yocto_dist/poky/meta-yocto-bsp \
  "
METADIR  := "${@os.path.abspath('/home/tce/yocto_dist/meta-updater/scripts/../..')}"
BBLAYERS += "${METADIR}/meta-updater"
BBLAYERS += "${METADIR}/meta-openembedded/meta-filesystems"
BBLAYERS += "${METADIR}/meta-openembedded/meta-oe"
BBLAYERS += "${METADIR}/meta-openembedded/meta-python"
BBLAYERS += "${METADIR}/meta-updater-raspberrypi"
BBLAYERS += "${METADIR}/meta-raspberrypi"
BBLAYERS += "${METADIR}/meta-openembedded/meta-networking"
BBLAYERS += "${METADIR}/meta-blinkenlights"
BBLAYERS += "/home/tce/yocto_dist/build/workspace"
BBLAYERS += "/home/tce/yocto_dist/meta-security/meta-tpm"
BBLAYERS += "/home/tce/yocto_dist/meta-slb9670-rpi"
```

local.conf
```
MACHINE ??= "qemux86"

DISTRO ?= "poky"

PACKAGE_CLASSES ?= "package_rpm"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

USER_CLASSES ?= "buildstats image-mklibs image-prelink"

PATCHRESOLVE = "noop"

BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
#ASSUME_PROVIDED += "libsdl2-native"

# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"

MACHINE = "raspberrypi3"
DISTRO = "poky-sota-systemd"

# General SOTA setup #############

# OTA Connect
SOTA_CLIENT_PROV = "aktualizr-shared-prov"
SOTA_PACKED_CREDENTIALS = "/home/tce/Downloads/new_credentials/credentials.zip"

CORE_IMAGE_EXTRA_INSTALL += " openssh "

# Uncomment this line to set the log level of aktualizr to 'debug' (from 'info'
# by default)
IMAGE_INSTALL_append += " aktualizr-log-debug"
# Store systemd logs in persistent storage
#
# It greatly helps diagnosing issues on testing devices but should be
# carefully weighted against file system usage and flash device wear for
# production systems. Please refer to systemd's docs for more details
IMAGE_INSTALL_append += " systemd-journald-persistent"

# Wifi configuration
#RPI_WIFI_ENABLE = "1"
#RPI_WIFI_SSID = "************"
#RPI_WIFI_PWD = "********"

# Automatically reboot after an update
SOTA_CLIENT_FEATURES = "ubootenv"
IMAGE_INSTALL_append = " aktualizr-auto-reboot "

# Make sure there is some space left
IMAGE_OVERHEAD_FACTOR = "2"

#IMAGE_INSTALL_append += " blinkenlights "
#OSTREE_COMMIT_BODY = "with blinkenlights"
INHERIT+="toaster buildhistory"

###########################################
# Add TPM support

ENABLE_UART = "1"
ENABLE_SPI_BUS = "1"
RPI_EXTRA_CONFIG = "dtoverlay=letstrust-tpm"

MACHINE_FEATURES_append = " tpm tpm2"

DISTRO_FEATURES_append = " systemd tpm tpm2 "
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

###########################################
IMAGE_INSTALL_append = " tpm2-tools libtss2 libtss2-tcti-device libtss2-tcti-mssim tpm2-abrmd tpm2-pkcs11 slb9670"
KERNEL_DEVICETREE_append = " overlays/letstrust-tpm.dtbo"
CORE_IMAGE_EXTRA_INSTALL = " kernel-modules "

```
I removed most of the comments to keep it concise

clem844, 2020-03-17:
I've removed all the TPM related part at the end of local.conf and have this error
```
ERROR: volatile-binds-1.0-r0 do_install: Execution of '/home/tce/yocto_dist/build/tmp/work/all-poky-linux/volatile-binds/1.0-r0/temp/run.do_install.31670' failed with exit code 1:
install: cannot stat 'mount-copybind': No such file or directory
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/tce/yocto_dist/build/tmp/work/all-poky-linux/volatile-binds/1.0-r0/temp/log.do_install.31670
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/tce/yocto_dist/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/tce/yocto_dist/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/tce/yocto_dist/poky/meta/recipes-core/systemd/systemd-systemctl-native.bb:do_populate_sysroot', '/home/tce/yocto_dist/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['patch-native', 'pseudo-native', 'systemd-systemctl-native', 'quilt-native', 'attr-native', 'gnu-config-native', 'automake-native', 'libtool-native', 'autoconf-native', 'gettext-minimal-native', 'texinfo-dummy-native', 'm4-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| install: cannot stat 'mount-copybind': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/tce/yocto_dist/build/tmp/work/all-poky-linux/volatile-binds/1.0-r0/temp/run.do_install.31670' failed with exit code 1:
| install: cannot stat 'mount-copybind': No such file or directory
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/tce/yocto_dist/poky/meta/recipes-core/volatile-binds/volatile-binds.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1749 tasks of which 1744 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds

Summary: 1 task failed:
  /home/tce/yocto_dist/poky/meta/recipes-core/volatile-binds/volatile-binds.bb:do_install
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

clem844, 2020-03-17:
That last problem with volatile-binds is solved.
However I still have the same error as in the beginning, even with the TPM part removed...
```
Error: 
 Problem: package packagegroup-base-extended-1.0-r83.raspberrypi3 requires packagegroup-base, but none of the providers can be installed
  - package packagegroup-base-1.0-r83.raspberrypi3 requires packagegroup-base-wifi, but none of the providers can be installed
  - conflicting requests
  - nothing provides wireless-regdb-static needed by packagegroup-base-wifi-1.0-r83.raspberrypi3
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

clem844, 2020-03-19:
Ok, I started from scratch and core-image-base builds successfully. The problem comes with poky-sota-systemd distro. No link with meta-raspberry then. I will be looking at meta-updater more closely.
This issue can be closed.

agherzan, 2020-03-19:
Thanks for the feedback. Closing.

closed 2020-03-19 by agherzan

kadapppa, 2023-04-30:
For me also, do_rootfs is getting failed for rpi-test-image