agherzan/meta-raspberrypi #589: RPI_EXTRA_CONFIG += in local.conf overwrites arm_64bit=1

m5p3nc3r, 2020-03-06:
**Description**

When I add
RPI_EXTRA_CONFIG += "\ndtoverlay=pi3-disable-bt"

to my local.conf for a 64bit RPI4 build it overwrites the (required?) addition of arm_64bit=1

This results in the image not being able to boot.

**Steps to reproduce the issue:**
1. Add RPI_EXTRA_CONFIG += "anything you want to add.." to your local.conf

**Describe the results you received:**
Does not boot

**Describe the results you expected:**
Boots

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

I have fixed this locally by updating conf/machine/raspberrypi4-64.conf to change the line RPI_EXTRA_CONFIG ?= "..."  to RPI_EXTRA_CONFIG = "..."

Not sure if this is the correct way to do it.  Will supply a patch for this if needed?

agherzan, 2020-03-06:
This happens because the local.conf is evaluated before the machine.conf. Try "_append" as opposed to "+=".

m5p3nc3r, 2020-03-06:
I understand this works, but I don't think its very intuitive?

The arm_64bit=1 is required for the boot, not optional as far as I can make out.  So why not make it mandatory to fix the majority of use-cases, and then it can be removed (for those who know what they are doing) by using "_remove"?  This way, it is still a standard process for adding to the config.txt in my local.conf by using the "+=" method.

agherzan, 2020-03-06:
That configuration is currently a workaround due to firmware issues. Follow the link provided in the configuration file to get more context. `RPI_EXTRA_CONFIG` is meant to be a generic mechanism so it doesn't enforce any defaults. I agree this is a bit unintuitively but keeps the current primitives. I guess another way to do it would be to force this configuration from the config recipe. I'm open to proposals.

m5p3nc3r, 2020-03-06:
So this should be a transient problem for now?  If so, I'm probably the only person who has fallen over this?  Maybe we can just ignore this, or add something to the documentation to warn against the overwriting behaviour.

The proposal I would make would be to change it from a ?= to a += in conf/machine/xxx.conf (until the firmware is fixed).  This way adding a += to your local.conf will not override the default setting.  And if you really have to remove it, you could use a '_remove' in your local.conf.

I'm not an OE expert, so I'm not sure if that will work, but it works for me :¬)

agherzan, 2020-03-06:
`+=` also works. I am for that change too.

m5p3nc3r, 2020-03-11:
Ok, when I get some time I will get a PR together.

prudhvi-cmx, 2021-01-07:
Hello, is the PR out for this ? I am facing the same issue

m5p3nc3r, 2021-01-08:
I didn't create a PR for this as the problem was supposed to be transient.  If the problem still exists, the solution is described in this thread.

agherzan, 2021-01-12:
I was under the impression you were going to put a PR together. @m5p3nc3r To make it an append as opposed to a weak assignment.

m5p3nc3r, 2021-01-12:
@agherzan I can pull something together this week.  I was under the impression from your comments earlier in the thread that this would be fixed in firmware so would be a transient fix in the recipes?

If this position has changed, I'll submit the PR later this week.

m5p3nc3r, 2021-01-14:
Just created #792