agherzan/meta-raspberrypi #591: Floating point performance is not good as raspbian

andung07, 2020-03-08:
**Description**

On raspberrypi-4 32bit yocto build, floating point calculation takes longer (double) than on official raspbian. A simple benchmarking c++ code is cross compiled on amd64 with hard floating point enabled. The same binary is transferred to poky build and raspbian.

**Steps to reproduce the issue:**
1. Compile the code:
```cpp
auto start = std::chrono::system_clock::now(); 
auto value = 0.0;
for ( int i = 0; i < 99999; i++) {
    value += 1.2123 * 0.3123;
}
auto end = std::chrono::system_clock::now();
std::cout << (uint64) std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count() << 
 "ms" << std::endl;
```
2. enable hard float compilation
```cmake
SET ( CMAKE_C_COMPILER "arm-linux-gnueabihf-gcc" )
SET ( CMAKE_CXX_COMPILER "arm-linux-gnueabihf-g++" )
SET ( CMAKE_CXX_FLAGS "-std=c++11 -march=armv8-a -mfloat-abi=hard -mfpu=neon-vfpv4 -fopenmp ${CMAKE_CXX_FLAGS}" )
```
3. scp the binary to raspbian and yocto build on raspberrypi4.

**Describe the results you received:**
Raspbian: 7 ms
Yocto build: 17 ms

**Describe the results you expected:**
The same performance on yocto build.

**Additional information you deem important (e.g. issue happens only occasionally):**
I believe the 32 bit yocto build for meta-raspberrypi4 is already using hard float tuning. Do you know what other optimization on raspbian that can be implemented in meta-raspberrypi in order to speed up the floating point operation?

kraj, 2020-03-08:
I have pi3 handy, so I tried this example and it consistently runs within 2ms for me ( using OE master )

andung07, 2020-03-09:
@kraj  thanks for testing that on raspi3. I updated the example and it gets more obvious if the loop is longer.

andung07, 2020-03-09:
It turned out the default cpu scalling governor is "powersave", thus the cpu clock is kept to minimum (600 MHz). After set the policy to "ondemand" or "performance", it gave the same performance as raspbian. 

`
echo "ondemand" > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
`

closed 2020-03-09 by andung07

kraj, 2020-03-09:
@andung07  question is that is this setting disabled in our kernel by default ?
if so then we should fix it

agherzan, 2020-03-09:
I don't think it's disabled by I do think that the default governor right now is powersave.

kraj, 2020-03-09:
@agherzan I think we should have a way to change that or document it otherwise we will see these kind of benchmarks showing Yoct/meta-rpi in bad light.

agherzan, 2020-03-09:
What's the default in Raspbian? @andung07  is it "ondemand"?

andung07, 2020-03-10:
@kraj No, it is not disabled in our kernel. All governor modes are available, but the default one is "powersave". When I inspect the kernel configuration of these OSes, both are using "powersave" by default.

`CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE=y`

@agherzan In raspbian, it is set at boot to "ondemand" by a systemd service raspi-config.service if the Shift key is not pressed during booting. Otherwise, it will be the same with us to fall down in "powersave" (based on the kernel default).

I think it is worth mentioning this in our documentation, so that users who expect to maximize the raspi capability are aware of this setting.

agherzan, 2020-03-10:
I was bitten in the past by this - now that I think about it. I reckon it is sensible to start defaulting the default to ondemand. @andung07 can you point me to the `raspi-config.service` file? I'm curious about what they do there.

andung07, 2020-03-11:
@agherzan my bad, it is just `raspi-config` file resides in `/etc/init.d/raspi-config`.

```bash
#!/bin/sh
### BEGIN INIT INFO
# Provides:          raspi-config
# Required-Start: udev mountkernfs $remote_fs
# Required-Stop:
# Default-Start: S 2 3 4 5
# Default-Stop:
# Short-Description: Switch to ondemand cpu governor (unless shift key is pressed)
# Description:
### END INIT INFO

. /lib/lsb/init-functions

case "$1" in
  start)
    log_daemon_msg "Checking if shift key is held down"
    if [ -x /usr/sbin/thd ] && timeout 1 thd --dump /dev/input/event* | grep -q "LEFTSHIFT\|RIGHTSHIFT"; then
      printf " Yes. Not enabling ondemand scaling governor"
      log_end_msg 0
    else
      printf " No. Switching to ondemand scaling governor"
      SYS_CPUFREQ_GOVERNOR=/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
      if [ -e $SYS_CPUFREQ_GOVERNOR ]; then
        echo "ondemand" > $SYS_CPUFREQ_GOVERNOR
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
        echo 100000 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor
      fi
      log_end_msg 0
    fi
    ;;
  stop)
    ;;
  restart)
    ;;
  force-reload)
    ;;
  *)
    echo "Usage: $0 start" >&2
    exit 3
    ;;
esac

```

agherzan, 2020-03-21:
That makes sense now. I truly believe that this should be the default. @andung07 do you feel like pushing a patch for enabling this by default in the kernel config?