agherzan/meta-raspberrypi #593: perf_event_open fails on rpi4

kraj, 2020-03-11:
strace has serveral failures which all lead to one problem that is __NR_perf_event_open syscall fails reliably on rpi4/32-bit kernel. Here is small example which should work fine if PMU was enabled and functioning correctly

```c
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>
#include <string.h>
#include <sys/ioctl.h>
#include <linux/perf_event.h>
#include <asm/unistd.h>

static long
perf_event_open(struct perf_event_attr *hw_event, pid_t pid,
int cpu, int group_fd, unsigned long flags)
{
int ret;

ret = syscall(__NR_perf_event_open, hw_event, pid, cpu,
group_fd, flags);
return ret;
}

int
main(int argc, char **argv)
{
struct perf_event_attr pe;
long long count;
int fd;

memset(&pe, 0, sizeof(struct perf_event_attr));
pe.type = PERF_TYPE_HARDWARE;
pe.size = sizeof(struct perf_event_attr);
pe.config = PERF_COUNT_HW_INSTRUCTIONS;
pe.disabled = 1;
pe.exclude_kernel = 1;
pe.exclude_hv = 1;

fd = perf_event_open(&pe, 0, -1, -1, 0);
if (fd == -1) {
fprintf(stderr, "Error opening leader %llx\n", pe.config);
exit(EXIT_FAILURE);
}

ioctl(fd, PERF_EVENT_IOC_RESET, 0);
ioctl(fd, PERF_EVENT_IOC_ENABLE, 0);

printf("Measuring instruction count for this printf\n");

ioctl(fd, PERF_EVENT_IOC_DISABLE, 0);
read(fd, &count, sizeof(long long));

printf("Used %lld instructions\n", count);

close(fd);
}
```

This however returns `Error opening leader 1`

a working system returns
```
Measuring instruction count for this printf
Used 76584 instructions
```

agherzan, 2020-03-12:
Interesting. Is this the case on Raspbian too?

kraj, 2020-03-12:
I dont know, since I dont have raspbian handy, but someone with one can try it out.

agherzan, 2021-10-30:
This was fixed a long time ago. Closing.

closed 2021-10-30 by agherzan