agherzan/meta-raspberrypi #598: Question: "now what?"

cristiandgal, 2020-03-18:
Hey guys, this isn't an issue, more of a 'now what' question.
Got the RPI booted, but only port 22 and 111 are open, there's no 443 (for REST) or UDP for IPMI.
Any pointers? Thanks.

agherzan, 2020-03-18:
Your message is extremely confusing. What are you trying to achieve?

cristiandgal, 2020-03-18:
Yeah .. sorry, I was a little too in my head. I'm trying to install OpenBMC on rpi3. Got the build per your docs, but it has no OpenBMC code, so was wondering, after I get the rpi running, where do I go to install the OpenBMC code. It's not very clear from the doc. Thanks.

agherzan, 2020-03-19:
Have you seen https://github.com/openbmc/openbmc? I would redirect any questions to them.

closed 2020-03-19 by agherzan