agherzan/meta-raspberrypi #600: RPI3 official touchscreen / HDMI is not working

clementlgl, 2020-03-26:
I'm trying to work with Official Raspberry PI 7" Touchscreen and the latest branch (Zeus). When I build `rpi-basic-image` with rocko, the psplash is displayed. But the same image on Zeus, the touchscreen display just the 4 raspberry and the kernel boot.

I have connected a display on HDMI and I have the same result ...

I work with raspberry Pi 3

My conf/local.conf
```
MACHINE = "raspberrypi3"
DISTRO = "poky"
PACKAGE_CLASSES = "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K \
"

PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"

DL_DIR ?= "/home/bionic/claigle/Dev/Yocto/download"
```

Have you the same problem ?

Thank you

texierp, 2020-03-26:
Hi @clementlgl ,

It seems you have the same problem as https://github.com/agherzan/meta-raspberrypi/issues/500

I think the variable `DISABLE_VC4GRAPHICS = "1"` solves your issue.

FYI, here is my configuration for the rpi3 and it works fine:
  - https://github.com/texierp/kas-files/blob/master/board-raspberrypi3.yml

PJ

agherzan, 2020-04-03:
I don't have a screen to test. It would be nice to see if this happens on rPi4 as well.

kraj, 2020-04-03:
I dont test zeus but on master pi4/vc4graphics work fine with weston/wayland and pi3 works ok with eglfs/userland graphics, I did see it stuck with boot logo with vc4graphics

clementlgl, 2020-04-06:
Thank you for your answers. 

Indeed, with `DISABLE_VC4GRAPHICS = "1"` it works ! But with `sysvinit` the psplash do not work, I switched to systemd.

closed 2020-04-06 by clementlgl