agherzan/meta-raspberrypi #605: Raspberry Pi Zero W Console No Output with ENABLE_UART = "1"

phishstang65, 2020-03-29:
**Description**

**Steps to reproduce the issue:**
1. Set MACHINE ??= "raspberrypi0-wifi"
2. Add ENABLE_UART = "1" to local.conf
3. Build core-image-minimal
4. Flash chard and boot

**Describe the results you received:**
No Console Output to Mac Terminal screen command

**Describe the results you expected:**
Console Output to Mac Terminal screen command

**Additional information you deem important (e.g. issue happens only occasionally):**
This has never worked for me. 

**Additional details (revisions used, host distro, etc.):**
Host is a Macbook Pro

agherzan, 2020-03-30:
I don't have a 0w to test but it would help if you could run, as a sanity check, a raspbian image.

phishstang65, 2020-03-30:
Thanks for the comment. It turns out that I get console output on my native Ubuntu machine - but not my MacBook, maybe something with the Prolific USB/Serial drivers there.. Anyway this is not a bug with meta-raspberry and so I'll close it.

closed 2020-03-30 by phishstang65