agherzan/meta-raspberrypi #608: No splash screen on raspberry pi3

timemaster5, 2020-03-31:
**Description**

No splash screen at all

**Steps to reproduce the issue:**
1. git clone poky, branch zeus
2. git clone meta-raspberrypi
3. set MACHINE to "raspberrypi3"
4. add meta-raspberrypi to bblayers.conf
5. source ...
6. bitbake core-image-base

**Describe the results you received:**
Splash screen with RPI logo shown, based on information found online, like this one https://github.com/agherzan/meta-raspberrypi/issues/320#issuecomment-429671791

**Describe the results you expected:**
No logo at all, not even Yocto splash :(

**Additional information you deem important (e.g. issue happens only occasionally):**
using ENABLE_UART="0" and SERIAL_CONSOLES="" to enable local HDMI screen instead of serial console

**Additional details (revisions used, host distro, etc.):**
tested with zeus and master branches
nothing else than bare minium in bblayers
also tested multiple images (rpi-basic-image, core-image-minimal, and so on..)

timemaster5, 2020-04-02:
Update:
Tested with systemd and sysvinit, no results.

But it seems the problem is in openembedded-core, if I upgrade openembedded-core to master from zeus, it works at least on systemd.

texierp, 2020-04-02:
Hi @timemaster5 

> But it seems the problem is in openembedded-core, if I upgrade openembedded-core to master from zeus, it works at least on systemd.

Indeed, oe-core now support psplash with systemd on master [1], that's why it's working, in fact zeus does not support systemd [2]. So you have to create your own systemd services (as done in [3]) if you want systemd support on this branch ;).

however, have no idea regarding sysvinit (not used on my side).

[1] - https://github.com/openembedded/openembedded-core/commit/b4063b16082604554c7c19b369ebddd27061f372#diff-af5cf4f300bd2b3fe46751a2fd54ccc0
[2] - https://github.com/openembedded/openembedded-core/blob/zeus/meta/recipes-core/psplash/psplash_git.bb#L118
[3] - https://git.automotivelinux.org/AGL/meta-agl/tree/meta-agl-profile-core/recipes-core/psplash/psplash_git.bbappend

timemaster5, 2020-04-02:
Thank you for this @texierp for this!

I wanted to report this behaviour because the implementation seems to be there and should work out of the box at least with sysvinit according to the information I found online, but it doesn't for some reason. 

Systemd is perfectly fine for me, thank you I will add the necessary changes to my layer.

closed 2020-04-02 by timemaster5