agherzan/meta-raspberrypi #613: Bootloader Optimization for meta-raspberrypi

mozcelikors, 2020-04-07:
Hi,

I am trying to reduce boot time on a project of mine that is based on Raspberry Pi 4 (meta-raspberrypi thud branch).
I have discovered that bootloader takes a huge time to load (~7seconds), surpassing kernel+userspace combined on my system. And I have recently noticed U-Boot is being ported (or have already been ported?) to this layer.  I am using the following HEAD:
`meta-raspberrypi     = "master:866ccc829f728f722f623571b24305d2bc9d8b17"`
Is there any chance you could confirm this HEAD contains all the necessities to switch to U-Boot? If so, briefly how do I do it?
Additionaly, what else would you recommend to reduce bootloader time on my system?

Thanks.

kraj, 2020-04-07:
you might set following in `local.conf`  to enable u-boot
```
RPI_USE_U_BOOT = "1"
```

timemaster5, 2020-04-07:
I use u-boot and confirm that it works well using the configuration option mentioned by @kraj

agherzan, 2020-04-08:
@mozcelikors , do you get a better boot time with uboot without changing anything else?

mozcelikors, 2020-04-08:
@agherzan Unfortunately not, it is nearly the same by looking at when kernel starts, although I have not yet been successful to fully measure it. My kernel+rootfs is around 2GB at the moment.
Still, I think there needs to be some optimization done somewhere, I am not knowledgeable exactly how raspberry pi boots, maybe I can start by rpi-eeprom optimization. I see a plan of integrating rpi-eeprom-update into meta-raspberrypi, when is it due?
Do you have other suggestions how I can optimize raspberry pi boot until at least kernel?
Thanks

agherzan, 2020-04-08:
That's why I was asking because I was suspecting no impact (at least not in the positive sense). Start to have some clear measures and figure out how much time things take. It will not be trivial to do it until kernel because you don't have any control (besides the txt configuration) of the RPi bootloader. The board will be using that with or without uboot. I'm interested in the outcome of your investigation but it sounds like something for which we will need some support from the foundation/community.

mozcelikors, 2020-04-08:
@agherzan Glad to hear you are interested. 

Regarding the bootloader,
My first idea is getting UART debug to work in bootloader, for that I need to update rpieeprom with DEBUG_UART variable. So far I have integrated rpi-eeprom-update to my layer (only that recipe and nothing else), however I get "VCHI Initialization failed error". So I am suspecting VideoCore is not fully loaded in meta-raspberrypi, I will continue to first integrate /opt/vc using online recipes then proceed. If you can think of quicker/better ways to get this working, I am always open for ideas. 

Regarding your further notes,
Do you also get such slow times for bootloader portion? I am just curious because my system have features such as sound/bluetooth/wifi/display which will slow things down, however I have not tried a vanilla meta-raspberrpi / Rasbian for a long time. If it is also taking long, at least until the kernel, then as you said we might ask foundation/community how this can be improved.

Regarding kernel improvements,
I think kernel provided by meta-raspberrypi is already pretty trimmed (at least non-necessary features are not included). Am I correct in this? Still there are printks/debugs+DWC OTG that I dont use that I took off, but that gives me 2.5 seconds. Any ideas on what else specifically could be trimmed for kernel config?

Thanks.

mozcelikors, 2020-05-30:
@agherzan Hi,
I am writing here as a follow up to my following question: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=275577&p=1669568

In short, in the Yocto-generated image, I get "VCHI Initialization failed" error whenever I use vcgencmd. This is being used in rpi-eeprom-update, therefore I need to figure this out in order to port rpi-eeprom-update to the meta-raspberrypi layer.

Looking at the kernel I notice there is no CONFIG_BCM_VIDEOCORE defined or included. I am also unable to do CONFIG_STAGING=y and CONFIG_BCM2835_VCHIQ=y. Could you please explain if this module is included in the kernel and configuration that meta-raspberrypi uses, what is different between meta-raspberrypi and Raspbian in terms of kernel perspective?

Thank you very much.

kraj, 2020-06-01:
@mozcelikors  can you use `DISABLE_VC4GRAPHICS = "1"` and see if that helps ?

closed 2021-10-30 by agherzan