agherzan/meta-raspberrypi #614: Configuration option DISABLE_RPI_BOOT_LOGO = "1" doesn't have any effect

timemaster5, 2020-04-07:
**Description**
It seems like the option of DISABLE_RPI_BOOT_LOGO = "1" doesn't do anything at least in the current master.  I wasn't able to find DISABLE_RPI_BOOT_LOGO text anywhere in the tree and resulting config.txt leaves the option disable_splash=1 untouched.

**Steps to reproduce the issue:**
1. set DISABLE_RPI_BOOT_LOGO = "1"
2. build the image

**Describe the results you received:**
Rainbow splash screen at the beginning of the boot process

**Describe the results you expected:**
No rainbow splash screen at the beginning of the boot process

**Additional details (revisions used, host distro, etc.):**
poky:zeus
meta-raspberrypi:master

texierp, 2020-04-07:
Hello @timemaster5 ,

To disable the raimbow screen, please use the following configuration:

```
# Disable raimbow screen
DISABLE_SPLASH = "1"
```

You can read more at [1].

The option `DISABLE_RPI_BOOT_LOGO` disable rpi boot logo from the command line [2] (Linux kernel side).

Thanks !

[1] - https://github.com/agherzan/meta-raspberrypi/blob/master/docs/extra-build-config.md#disable-splash-screen
[2] - https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi.inc#L36

timemaster5, 2020-04-07:
Oh sorry for that and thank you @texierp for pointing me out to the correct option. I was following this documentation: https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#disable-rpi-boot-logo

texierp, 2020-04-07:
> Oh sorry for that and thank you @texierp for pointing me out to the correct option

with pleasure

> which is probably outdated then.. but it is linked in the master README file

This is the same in fact ;)

The option is listed here : https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#disable-splash-screen

timemaster5, 2020-04-07:
Oh sorry again for my confusion... this is a different option at all. What is the option of DISABLE_RPI_BOOT_LOGO does then? Disable the kernel image which represents a CPU cores count?

texierp, 2020-04-07:
> Oh sorry again for my confusion... this is a different option at al

No problem ;)

> Disable the kernel image which represents a CPU cores count?

Yes, exactly !

closed 2020-04-07 by timemaster5

agherzan, 2020-04-08:
Thanks @texierp .

texierp, 2020-04-08:
with pleasure @agherzan ;)