agherzan/meta-raspberrypi #618: Hardware acceleration RPI4 FFmpeg

honcharenko-roman, 2020-04-09:
I'm trying to get to work hardware acceleration on my Raspberry pi 4-64. I'm using FFmpeg and AFAIK hwaccel can be reached by using OpenMAX or V4L2-M2M. 
After '--enable-omx' and 'enable-omx-rpi' for FFmpeg, build fails with error: 'OMX_Core.h not found'. If I will provide manually omx headers, then it will compile but FFmpeg encoding will fail due to missing libraries: bcm_host.so and libopenmaxil.so 
I have tried reverting to userland, it produced bcm_host.so, but not libopenmaxil.so. I have tried different combinations of virtual providers but without success. 
Is it possible to access omx hardware acceleration on RPI4-64?

**Steps to reproduce the issue:**
1. Download latest Poky distro, meta-openembedded, meta-raspberrypi
2. Enable omx, omx-rpi support for FFmpeg
3. Link headers for FFmpeg
4. Build and try to use h264_omx

How do I get missing library libopenmaxil.so and everything I need for hwaccel? 

poky master: commit 5d47cdf448b6cff5bb7cc5b0ba0426b8235ec478
meta-openembedded master: commit daa50331352c1f75da3a8ef6458ae3ddf94ef863
meta-raspberrypi master: commit 8d163dd844c3737d1557395c1eb512ebac86bdc0

BTW, by using V4L2-M2M, I'm getting green shadows on the resulting video. Maybe can someone point me in the right direction?

agherzan, 2020-04-15:
Userland doesn't support 64bit compilation - see https://github.com/raspberrypi/userland/issues/314. Maybe using precompiled binaries would work but that's a route we haven't tested it for a while (at least I didn't).

honcharenko-roman, 2020-04-16:
Thanks, I built a 32-bit version and now I'm able to use hardware encoding. Thanks.

closed 2020-04-16 by honcharenko-roman