agherzan/meta-raspberrypi #622: Hardware acceleration with VLC

lesti, 2020-04-12:
**Description**
I'm trying to get hardware acceleration in VLC via mmal to work (RPi 3B+). On the Pi it compiles just fine, but when cross-compiling with bitbake on my host, it seems there are some libraries missing. I'm trying to achieve hardware accelerated playback of h264 Video with VLC, but there is no GUI or similar needed. Omxplayer is not really usable for my project, since there is pre-buffering of following videos needed. 

**Steps to reproduce the issue:**
1. get the latest `poky`, `meta-raspberrypi` and `meta-openembedded`
2. set `--enable-mmal`, add `userland` to dependencies , remove unneeded packages and set flags ([my .bbappend here)](https://gist.github.com/lesti/b099c8da13845f395ef1572e89e39940)
3. in `local.conf` make sure to add vlc to `IMAGE_INSTALL`, and set `PREFERRED_PROVIDER_virtual/libgles2` and `PREFERRED_PROVIDER_virtual/egl` to `userland`. Also set `DISABLE_VC4GRAPHICS` to `1`
4. try to build vlc / core-image-minimal (this ist what I'm using)

**Describe the results you received:**
When compiling, everything runs just fine until `vlc` -> `do_configure`:
```
| checking for interface/mmal/mmal.h... yes
| checking for vc_tv_unregister_callback_full in -lbcm_host... no
| configure: error: Cannot find bcm library...
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/lesti/yocto/meta-openembedded/meta-multimedia/recipes-multimedia/vlc/vlc_3.0.8.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2749 tasks of which 2744 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/lesti/yocto/meta-openembedded/meta-multimedia/recipes-multimedia/vlc/vlc_3.0.8.bb:do_configure
```
So if I'm correct, it's missing the bcm_host.h file, which should be included with the `userland` package.
**Describe the results you expected:**
VLC / the image compiles fine and when running `vlc --list | grep vout` there shows up the `mmal_vout` option

**Additional details (revisions used, host distro, etc.):**
My build host is a freshly installed and updated Debian Buster.
```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:5d47cdf448b6cff5bb7cc5b0ba0426b8235ec478"
meta-raspberrypi     = "master:8d163dd844c3737d1557395c1eb512ebac86bdc0"
meta-oe              
meta-multimedia      
meta-networking      
```

agherzan, 2020-04-15:
It's worth looking into the userland package a bit looking for bcm_host. Do you get that compiled?

kraj, 2020-04-15:
`libbcm_host.so` is available with latest userland but I wonder what test vlc configure is doing. Can you paste config.log from build dir and perhaps that will have more info.

lesti, 2020-04-15:
> It's worth looking into the userland package a bit looking for bcm_host. Do you get that compiled?

If I add it to the `IMGAGE_INSTALL_append` in `local.conf` and leave out vlc this time, userland is compiled just fine into the image and the `libbcm_host.so` also gets created at `/usr/lib`. 
In the `userland` package provided by `meta-raspberrypi`, the userland package from the raspberrypi repository is used and includes bcm_host (see [here](https://github.com/raspberrypi/userland/tree/master/host_applications/linux/libs/bcm_host)). 
Also the [patch](https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-graphics/userland/files/0004-wayland-egl-Add-bcm_host-to-dependencies.patch) added by @kraj seems to have something to do with that library.

> `libbcm_host.so` is available with latest userland but I wonder what test vlc configure is doing. Can you paste config.log from build dir and perhaps that will have more info.

You're right, it is included with userland. I've uploaded my `config.log` of vlc (version 3.0.8) in [this gist](https://gist.github.com/lesti/deb396249d205e83daee0989becf22a4), but I can't find any obvious test in there (See the configure error at line 18575).

The `mmal_vout` module of VLC wants to include it (see [here](https://github.com/videolan/vlc/blob/master/modules/hw/mmal/vout.c#L35)), but I can't verify if that is what it's looking. Also in the [configure file of VLC](https://github.com/videolan/vlc/blob/master/configure.ac#L3708) it seems to check for the `bcm_host` lib.

kraj, 2020-04-15:
I think the problem is that vlc is assuming raspbian packaging structure of `/opt/vc` to these libs whereas in yocto we package them into `/usr/lib` and `/usr/include` see [here](https://github.com/videolan/vlc/blob/master/configure.ac#L3716)

ideally, they should have used packageconfig to enquire for these settings then it would have avoided hardcoding assumptions. 

So I guess we need to patch the above location so it is adding /usr/include into CPPFLAGS

kraj, 2020-04-16:
@lesti Please try the patch I posted for vlc [here](https://patchwork.openembedded.org/patch/171985/)
this will need you to be on latest master of meta-openembedded and your bbappend should look like
```
PACKAGECONFIG_remove = " \
        dc1394 \
        dv1394 \
        notify \
        freetype \
        dvdread \
        live555 \
        fontconfig \
        freetype \
        fluidsynth \
"

PACKAGECONFIG_append_rpi = " \
        x264 \
        mmal \
"
```

Let me know how it goes ....

lesti, 2020-04-16:
Ok, so I've updated meta-openembedded (vlc 3.0.9.2, yay!), applied that patch, updated my `vlc_%.bbappend` and now it compiles without complaining about the missing `bcm_host`, so that seems to have worked so far.
However, the `mmal_vout` module used to accelerate playback is still not present in the build, so I can't verify the mmal module is actually built into VLC and cannot use it (a `vlc --list | grep mmal` returns nothing)...

agherzan, 2021-11-13:
I expect that all the patches mentioned above have been now merged into the dependent layers, I'll close this issue for now.

closed 2021-11-13 by agherzan