agherzan/meta-raspberrypi #631: bitback stuck at linux-raspberrypi-1_4.19...

Xtazyatu, 2020-04-21:
**Description**
Every time I using bitbake i get stuck at the following line and always at 74% :
0: linux-raspberrypi-1_4.19.115+gitAUTOINC+b13fc60b52-r0 do_fetch (pid 65500)  74% |##############################################################                      | 18.24 MiB/s

I've searched the forums and it's not a matter of the task taking a long time, since i left it over night and when i woke up it did nothing. 

In all my tries i never managed to pass this step 

**Steps to reproduce the issue:**
1.  git clone -b master git://git.yoctoproject.org/poky
2. git clone -b master git://git.openembedded.org/meta-openembedded
3. git clone -b master https://github.com/agherzan/meta-raspberrypi.git
4. source oe-init-build-env
5. Added meta-raspberrypi to BBLAYERS:

```
  /home/elian/Yocto/poky/meta-raspberrypi \
  /home/elian/Yocto/poky/meta-openembedded/meta-filesystems \
  /home/elian/Yocto/poky/meta-openembedded/meta-gnome\
  /home/elian/Yocto/poky/meta-openembedded/meta-initramfs\
  /home/elian/Yocto/poky/meta-openembedded/meta-multimedia\
  /home/elian/Yocto/poky/meta-openembedded/meta-networking\
  /home/elian/Yocto/poky/meta-openembedded/meta-oe\
  /home/elian/Yocto/poky/meta-openembedded/meta-perl\
  /home/elian/Yocto/poky/meta-openembedded/meta-python\
  /home/elian/Yocto/poky/meta-openembedded/meta-webserver\
  /home/elian/Yocto/poky/meta-openembedded/meta-xfce\
```

6.  Added to conf/local.conf:

```
GPU_MEM = "16"

DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

IMAGE_INSTALL_append = " connman connman-client "

DISABLE_SPLASH = "1"

DISABLE_OVERSCAN = "1"
HDMI_GROUP="2"
HDMI_MODE="1"
HDMI_MODE="87"
HDMI_DRIVE="1"
HDMI_FORCE_HOTPLUG="1"
DISABLE_RPI_BOOT_LOGO = "1"
ENABLE_SPI_BUS = "1"
ENABLE_I2C = "1"
KERNEL_MODULE_AUTOLOAD_rpi += "i2c-dev i2c-bcm2708"

MACHINE_FEATURES += "pitft35r"
ENABLE_DWC2_PERIPHERAL = "1"

MACHINE_FEATURES_append = "vc4graphics"
DISTRO_FEATURES_append = " bluez5 bluetooth wifi polkit "

PACKAGECONFIG_append = " gstreamer"
LICENSE_FLAGS_WHITELIST = "commercial"
```

**Describe the results you received:**
I get stuck at the following line: 
Currently  8 running tasks (3838 of 4375)  87% |###################################################################################################################                 |
0: linux-raspberrypi-1_4.19.115+gitAUTOINC+b13fc60b52-r0 do_fetch (pid 65500)  74% |##############################################################                      | 18.24 MiB/s

**Describe the results you expected:**
After some time, the task should complete.

**Additional information you deem important (e.g. issue happens only occasionally):**
Run several times also left it overnight and didn't manage to get it working. 

**Additional details (revisions used, host distro, etc.):**
Distro='poky'

kraj, 2020-04-21:
It’s downloading git clone of kernel repository which is quite large (multiple gigabytes) so you might want to check you bandwidth and sometimes server side might disconnect due to glitches etc so please retry and see if that helps

Xtazyatu, 2020-04-21:
hi @kraj thanks for the quick reply. 

As mentioned in the bug report, i left it over night and also retried several times.
Is there a way to see what exacly it's downloading and were it get's stuck because i find it strange to get stuck every time at the same percentage 74%

Regards
Adrian.

kraj, 2020-04-21:
Can you clone it without yocto
`git clone https://github.com/raspberrypi/linux -b rpi-4.19.y`

agherzan, 2020-04-27:
@Xtazyatu Have you tried to clone the repository externally as suggested above? Does the issue still persist on your machine?

Xtazyatu, 2020-04-27:
Hi Andrei, Sorry for not posting a reply. Unfortunately, yes still stuck at 74%. Starting to think that this might be an issue with my configuration, although i don't see where could be there problem as it makes no sense to get stuck there everytime. 
I'll try a few more things and get back to you.

agherzan, 2020-04-27:
So it gets stuck on your machine by just doing git clone, did I understand you correctly?

Xtazyatu, 2020-04-27:
Hi Andrei, so i misread the question. Here is the correct answer. I can do a complete git clone as suggested by kraj. Everything is ok with git clone. Going back to the compiling part, i still get stuck at 74% as described above.

agherzan, 2020-04-27:
How long does a git clone on the same machine take?

Xtazyatu, 2020-04-27:
It takes about 10 minutes, here is the terminal output: 

> elian@ubuntu:~/rpiImage$ sudo git clone https://github.com/raspberrypi/linux -b rpi-4.19.y
> Cloning into 'linux'...
> remote: Enumerating objects: 7675, done.
> remote: Counting objects: 100% (7675/7675), done.
> remote: Compressing objects: 100% (3012/3012), done.
> remote: Total 7901591 (delta 5616), reused 5309 (delta 4653), pack-reused 7893916
> Receiving objects: 100% (7901591/7901591), 2.27 GiB | 19.62 MiB/s, done.
> Resolving deltas: 100% (6603051/6603051), done.
> Checking out files: 100% (62373/62373), done.

agherzan, 2020-06-09:
Have you had any luck with it?

Xtazyatu, 2020-06-25:
> Have you had any luck with it?

Hi Andrei, 
To be fare, i went with a totally different solution for what i wanted to do and dropped this completely. I can't answer your question as i haven't tried it since. 

Best Regards!

agherzan, 2020-06-25:
That's alright. I'll close this for now but feel free to comment if it persists.

closed 2020-06-25 by agherzan

matzoius, 2020-10-06:
Hi, I have the same problem.

kraj, 2020-10-06:
@matzoius can you clone kernel sources outside yocto using plain git clone command ?

matzoius, 2020-10-06:
Yes, separate cloning works smoothly. Here is the log:

sudo git clone https://github.com/raspberrypi/linux -b rpi-4.19.y linux_test
Cloning into 'linux_test'...
remote: Enumerating objects: 6981, done.
remote: Counting objects: 100% (6981/6981), done.
remote: Compressing objects: 100% (2637/2637), done.
remote: Total 8380604 (delta 5019), reused 5539 (delta 4343), pack-reused 8373623
Receiving objects: 100% (8380604/8380604), 2.43 GiB | 6.69 MiB/s, done.
Resolving deltas: 100% (6998543/6998543), done.
Checking objects: 100% (16777216/16777216), done.
Updating files: 100% (62376/62376), done.

In the image below is my bitbake console. the linux-raspberrypi task will remain forever at 74%
[image]

matzoius, 2020-10-06:
_bitbake linux-raspberrypi_ stops also at 74%
I tried witl all linux versions from poky-dunfell/meta-raspberrypi/recipes-kernel/linux and they all stop at 74%.

matzoius, 2020-10-06:
I can copy manually the git clone result on the build work folder to overcome this blocking point, but I don't know where the work folder for linux-raspberrypi is.

kraj, 2020-10-06:
I think fetcher is making a bare clone so it will take a bit longer I would say leave it for longer and see if it finishes

matzoius, 2020-10-06:
I found this on the do_fetch log:

Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|198.145.29.63|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-10-06 17:03:04 ERROR 404: Not Found.

DEBUG: Mirror fetch failure for url http://downloads.yoctoproject.org/mirror/sources/git2_github.com.raspberrypi.linux.git.tar.gz (original url: git://github.com/raspberrypi/linux.git;branch=rpi-4.19.y)

Is this s git mirror problem ?

kraj, 2020-10-06:
I doubt that, since its a pre-mirror and it should time out.

matzoius, 2020-10-07:
I left it over night, cpu calmed down since it arrived at 75% and it is still there.

kraj, 2020-10-07:
thats not looking good then. what happens if you do 
```
bitbake -ccleanall virtual/kernel
bitbake -cfetch virtual/kernel
```

does it finish?

matzoius, 2020-10-07:
First command:
_Initialising tasks: 100% |################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 0 (0% match, 0% complete)
NOTE: No setscene tasks
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 4 tasks of which 0 didn't need to be rerun and all succeeded._

Second command:
_Initialising tasks: 100% |##################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 0 (0% match, 0% complete)
NOTE: No setscene tasks
NOTE: Executing Tasks
Currently  1 running tasks (1 of 1)   0% |                                                                                                                |
0: linux-raspberrypi-1_4.19.126+gitAUTOINC+f6b3ac28f0-r0 do_fetch (pid 1817080)  74% |#########################################               | 13.72 MiB/s_

and remains stuck at 74%