agherzan/meta-raspberrypi #637: No sdimg created

JadElClemens, 2020-04-29:
**Description**

Building core-image-base and rpi-basic-image, I am unable to find any sdimg file generated in [build dir]/tmp/deploy/images/raspberrypi3-64/

**Steps to reproduce the issue:**
1. Clone poky, meta-openembedded, and meta-raspberrypi. 
2. Add the necessary packages to [build dir]/conf/bblayers.conf.
3. Change MACHINE in [build dir]/conf/local.conf to "raspberrypi3-64"
3. bitbake core-image-base or rpi-basic-image

**Describe the results you received:**

The build runs fine, but no sdimg file is generated. I tried to dd other files (Image-raspberrypi3-64.bin and core-image-base-raspberrypi3-64.ext3) to my SD card to no avail.

**Describe the results you expected:**
An sdimg file should be generated in accordance with the quickstart steps.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
Used master on all repos (poky, meta-oe, meta-rpi), running fedora30 in systemd-nspawn container on Arch.

rcla, 2020-05-02:
Earlier this year they changed the image type 'sdimg' to 'wic.bz2'

Bmaptool can be used to copy 'name-image.wic.bz2' to SD Card

Example: (Assuming image name is 'core-image-base ...' and SD Card is '/dev /mmcblk0')

Type inside the 'Deploy' folder:

`sudo bmaptool copy core-image-base-raspberrypi3-64.wic.bz2 /dev/mmcblk0`

Leistungsabfall, 2020-05-03:
To extend @rcla's answer a bit:

`bmaptool` can be installed on Ubuntu via apt:
`sudo apt install bmap-tools`

Make sure to unmount all SD card partitions before running bmaptool:
`umount /dev/sda1`
`umount /dev/sda2`

(Assuming SD card is `/dev/sda` with partitions `/dev/sda1` and `/dev/sda2`)

agherzan, 2020-05-09:
@JadElClemens thanks for this. This is indeed a documentation issue. Do you feel like pushing a commit to fix this?

JadElClemens, 2020-05-22:
Fixed in 4982da8, see #644

closed 2020-05-22 by JadElClemens