agherzan/meta-raspberrypi #639: bitbake rpi-test-image with MACHINE=raspberrypi3-64 failing to build on the zeus release of yocto

alexryan, 2020-04-30:
**Description**

bitbake rpi-test-image with MACHINE=raspberrypi3-64 failing to build on the zeus release of yocto

**Steps to reproduce the issue:**
Fully documented here:
https://stackoverflow.com/questions/61534170/bitbake-rpi-test-image-with-machine-raspberrypi3-64-failing-to-build-on-the-zeus

**Describe the results you received:**
Fully documented here:
https://stackoverflow.com/questions/61534170/bitbake-rpi-test-image-with-machine-raspberrypi3-64-failing-to-build-on-the-zeus

**Describe the results you expected:**
An image that I could dd to a microSD card.

**Additional information you deem important (e.g. issue happens only occasionally):**
General advice how how to debug these build errors when googling on the error message fails would be appreciated.

**Additional details (revisions used, host distro, etc.):**

agherzan, 2020-06-09:
I would appreciate also have a description here not only on StackOverflow.

The error is:
```
  - package wireless-regdb-2019.06.03-r0.noarch conflicts with wireless-regdb-static provided by wireless-regdb-static-2019.06.03-r0.noarch
```

There is a conflict in between the packageroup in meta-raspberrypi (see packagegroup-rpi-test.bb) and the packagegroup in base (poky - see packagegroup-base.bb). Poky brings in `wireless-regdb-static` while meta-raspberrypi brings in `wireless-regdb`. We will need to switch the rpi packagegroup to `wireless-regdb-static` as well. Can you push a PR?

jmagnuson, 2020-06-23:
I ran into this issue myself and submitted a fix in #664.

agherzan, 2020-06-24:
Thanks @jmagnuson . Merged to master.

PR #664 (2020-06-24): packagegroup-rpi-test: resolve `wireless-regdb` conflict
Replaces `wireless-regdb` with `wireless-regdb-static` provided by poky.

Fixes #639.
commit aa47ab63 (2020-06-24): packagegroup-rpi-test: resolve `wireless-regdb` conflict

Replaces `wireless-regdb` with `wireless-regdb-static` provided by poky.

Fixes #639.
commit ee402d3e (2020-07-14): packagegroup-rpi-test: resolve `wireless-regdb` conflict

Replaces `wireless-regdb` with `wireless-regdb-static` provided by poky.

Fixes #639.
closed 2020-06-24 by agherzan