agherzan/meta-raspberrypi #640: Console messages not printing properly thorugh UART

bhargavthriler, 2020-05-03:
**Description**

Console messages not printing properly thorugh UART while using `master` branch of meta-raspberrypi and poky
When `ENABLE_UART = "1"` and `RPI_USE_U_BOOT = "1"` is added in `local.conf`

But when switch back `zeus` in both the branches (meta-raspberrypi and poky) console works fine. Is there any new change after `zeus` ?

Or am I doing some mistake here ?

**Steps to reproduce the issue:**
1. Clone poky
2. Clone meta-raspberrypi
3. Add `ENABLE_UART = "1"` and `RPI_USE_U_BOOT = "1"` in `conf/local.conf` before building . 
4. Flash sd card using bmaptool
5. Connect gpio pin 6, 8, 10 (gnd, txd, rxd) to Usb To Rs232 Ttl Converter.
6. Use minicom to connect to console

**Describe the results you received:**
```
U-Boot 2020.04 (Apr 13 2020 - 15:02:18 +0000)

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@7e980000: scanning bus usb@7e980000 for devices... �
                                                            L��2`
                                                                 �br��`�|b
                                                                            ��~��|��rl

                                                                                      �|2�nn��n�L��B��~B�|��n|��
                                                                                                                     �br����bbb

                                                                                                                                 �l��n|��
                                                                                                                                            �br��`�|b
                                                                                                                                                       ��~��|��rl`2n�L�n��r2
                                                                                                                                                                               �~�|��nL����np~�n��

                                                                                                                                                                                                    �
                                                                                                                                                                                                      ```
                                                                                                                                                                                                         L`��n

```

**Describe the results you expected:**
While building from `zeus` branch
```
U-Boot SPL 2019.04-4.19.35-1.1.0+g4d37753 (Dec 23 2019 - 10:51:14 +0000)
power_bd71837_init
DDRINFO: start DRAM init
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
Trying to boot from ROMAPI
image offset 0x8000, pagesize 0x200, ivt offset 0x0

U-Boot 2019.04-4.19.35-1.1.0+g4d37753 (Dec 23 2019 - 10:51:14 +0000)

CPU:   Freescale i.MX8MNano Quad rev1.0 1500 MHz (running at 1200 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 38C
Reset cause: POR
Model: NXP i.MX8MNano EVK board
DRAM:  2 GiB
TCPC:  Vendor ID [0x1fc9], Product ID [0x5110], Addr [I2C1 0x52]
Power supply on USB2
TCPC:  Vendor ID [0x1fc9], Product ID [0x5110], Addr [I2C1 0x50]
MMC:   FSL_SDHC: 1, FSL_SDHC: 2
Loading Environment from MMC... Run CMD11 1.8V switch
*** Warning - bad CRC, using default environment

No panel detected: default to MIPI2HDMI
adv7535_init: Can't find device id=0x3d, on bus 1
Display: MIPI2HDMI (1920x1080)
Video: 1920x1080x24
In:    serial
Out:   serial
Err:   serial

 BuildInfo:
  - ATF 70fa7bc
  - U-Boot 2019.04-4.19.35-1.1.0+g4d37753

Run CMD11 1.8V switch
switch to partitions #0, OK
mmc1 is current device
flash target is MMC:1
Run CMD11 1.8V switch
Net:   
Warning: ethernet@30be0000 using MAC address from ROM
eth0: ethernet@30be0000
Fastboot: Normal
Normal Boot
Hit any key to stop autoboot:  0 
Run CMD11 1.8V switch
switch to partitions #0, OK
mmc1 is current device
Run CMD11 1.8V switch
24263168 bytes read in 387 ms (59.8 MiB/s)
Booting from mmc ...
40173 bytes read in 6 ms (6.4 MiB/s)
## Flattened Device Tree blob at 43000000

```

Leistungsabfall, 2020-05-03:
I have the same problem on a PI Zero W.

PatrickLang, 2020-05-03:
Same on rpi zero w. Thud worked, dunfell does not

PatrickLang, 2020-05-04:
Yes, something changed related to the GPU clock rate on the RPi Zero-W.

I manually added this to `/boot/config.txt`, and now the UART works for serial console:
```
gpu_freq=250
```

PatrickLang, 2020-05-04:
I tried adding these lines to `local.conf` and built an image but they didn't have any effect.

```
GPU_FREQ = "250"
GPU_MEM = "16"
```

PatrickLang, 2020-05-04:
`${DEPLOYDIR}/bcm2835-bootfiles/config.txt` has the right options in it

Leistungsabfall, 2020-05-04:
Thanks for sharing your workaround, works great!

kraj, 2020-05-04:
Does it work ok without u-boot ?

PatrickLang, 2020-05-04:
@kraj I am not using u-boot And it works when I set gpu clock

kraj, 2020-05-04:
your description states you are `RPI_USE_U_BOOT = "1"`

can you comment that out and see if you still have issues ?

bhargavthriler, 2020-05-05:
> your description states you are `RPI_USE_U_BOOT = "1"`
> 
> 
> 
> can you comment that out and see if you still have issues ?

Yes even without u-boot the issue is same

kraj, 2020-05-05:
Ok then I guess we need a patch to boot files recipe to change the gpu frequency settings

bhargavthriler, 2020-05-06:
Thanks @PatrickLang your works great with rpi3 b+ also

bhargavthriler, 2020-05-06:
> Ok then I guess we need a patch to boot files recipe to change the gpu frequency settings

Exactly what changed ? and where can I find the change if I want check the root cause of this issue ?

agherzan, 2020-06-09:
This sounds to me like a bug in upstream. Have we followed up? Does raspbian set the GPU freq as well?

bhargavthriler, 2021-01-29:
This is fixed in the latest build.
Closing this issue.

agherzan, 2021-11-13:
@bhargavthriler This seems to be an old issue. Are you still looking into it? If yes, can you please try to replicate with the latest master and provide feedback if this is reproducible on setups with u-boot and/or setups without uboot?

agherzan, 2022-02-11:
Closing due to inactivity. I'll close this issue but feel free to comment if we get more info.

closed 2022-02-11 by agherzan

bhargavthriler, 2022-02-12:
> @bhargavthriler This seems to be an old issue. Are you still looking into it? If yes, can you please try to replicate with the latest master and provide feedback if this is reproducible on setups with u-boot and/or setups without uboot?

I missed this, I will look into this.

bhargavthriler, 2022-02-14:
Issue resolved ( checked with latest build ).

agherzan, 2022-02-19:
> Issue resolved ( checked with latest build ).

Thanks for confirming.