agherzan/meta-raspberrypi #643: zeus branch raspberrypi cm3 u-boot no messages in console

nandra, 2020-05-05:
Using meta-raspberrypi branch zeus with raspberrypi cm3 module and enable u-boot in localconf. Build image doesn't show any info in console. I tried older u-boot 2018.01 and it works fine. Ideas? 
Console is on PIN 14 and 15.
Thanks.

agherzan, 2020-06-09:
Could it be related to https://github.com/raspberrypi/firmware/issues/1391 ?

nandra, 2020-06-09:
Issue was missgin u-boot patch: commit id: 29f7d05a347ab7a42577c67fdfb787ef91537302

agherzan, 2020-06-09:
Do we have that in meta-raspberrypi?

nandra, 2020-06-09:
Nope should be cherrypicke at least for have u-boot 2020.01 working on  cm3.

mike-scott, 2020-06-09:
@nandra Ah that makes sense.  I ended up migrating to a newer u-boot and my boot issues on CM3 went away.

@agherzan 
I notice that `https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/raspberrypi-cm3.conf` doesn't set:
`VC4DTBO ?= "vc4-fkms-v3d"` like the other machine configurations.

I had to temporarily setup an append to get working fkms graphics.

Otherwise the default setting here is set (kms):
https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/bootfiles/rpi-config_git.bb#L27

mike-scott, 2020-06-09:
To clarify: I know that this setting in config.txt shouldn't really affect the u-boot / Linux.  But I was getting a GPU hang during Linux boot until I made sure the config.txt had the right setting here.  Who knows what the RaspberryPi firmware does prior to u-boot handoff.

nandra, 2020-06-20:
Fix: #660

amgedr, 2020-09-22:
U-Boot was not working in dunfell for the CM3+. It was fixed when I manually added the changes in #660. Why was the commit reverted from the dunfell branch?

agherzan, 2020-09-22:
Can you point me to the revert PR?

amgedr, 2020-09-23:
Sorry, should've mentioned that. Its #699

nandra, 2020-09-25:
Also stock zeus build (without u-boot) doesn't boot on CM3. Anyone have some idea? CM3 is light version without eMMC. Thanks.

nandra, 2020-09-25:
Nah I was wrong. It just took some time until it start and need to enable console :). Sorry for noise.

agherzan, 2020-10-05:
@amgedr That PR was a hefty backport. Would be hard for me to spot the issue from it. We would need to do a bit of investigation.

amgedr, 2020-10-08:
Hope this helps when you get a chance to review: #660's commit f20ccfa11db4c0d1608cf6f3e654d29c5bfc3dd3 solved the CM3+ booting issue when u-boot is enabled. But as you commented in #699, it was reverted in 8258e733ea89b6ac1f65419a8baf42110b83fdbc because it was missing the following line:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
```

agherzan, 2020-10-14:
That makes sense - and I remember now. Do you feel like pushing that? CC @nandra

nandra, 2020-10-14:
Should I redo patch or @amgedr will you do it? Thanks.

agherzan, 2020-10-14:
I didn't know you are up for it. If you want, it would probably be better if you could do it as you know better patch.

nandra, 2020-10-15:
PR #747

agherzan, 2020-10-16:
Thanks @nandra . @amgedr do you have a second to give the PR a spin?

amgedr, 2020-10-19:
Sorry, didn't notice the notification email earlier. Works on CM3+.

vkotarov, 2021-06-08:
Not sure if this is the right place to comment but the patch somehow made it to gatesgarth. On the other hand, it's not needed there as u-boot version is 2020.07 and already has it applied. Eventually, the do_patch task breaks on gatesgarth (probably on hardknott, too)

agherzan, 2021-10-30:
Looks fixed. Ping me otherwise.

closed 2021-10-30 by agherzan