agherzan/meta-raspberrypi #658: failure with with 5.4.45 linux-raspberrypi recipes

embdlover, 2020-06-18:
Hi,
i have built the image for raspberrypi4-64 with 4.X kernel.
Now i trying to build an image for 5.4 kernel,but i am trying to the change the kernel i am getting following error.
i have added the following line in the local.conf
**PREFERRED_VERSION_linux-raspberrypi="5.4.%"**

meta                 = "master:982b7f98b8423236cc986346379b1bde3694f131"
meta-raspberrypi     = "dunfell:8066fac91d4167504147886fe0caf429e7271466"
meta-oe              
meta-python          
meta-networking      
meta-multimedia      
meta-perl            = "dunfell:e413c1ef621688e69bb7830bb3151ed23b30b73e"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:07
Sstate summary: Wanted 28 Found 0 Missed 28 Current 2260 (0% match, 98% complete)
NOTE: Executing Tasks
ERROR: linux-raspberrypi-1_5.4.45+gitAUTOINC+d00cdd80ab-r0 do_patch: Could not apply patches for raspberrypi4-64.
ERROR: linux-raspberrypi-1_5.4.45+gitAUTOINC+d00cdd80ab-r0 do_patch: Patch failures can be resolved in the linux source directory /home/pankaj/Oe-Raspberrypi4/tmp-glibc/work-shared/raspberrypi4-64/kernel-source)
ERROR: linux-raspberrypi-1_5.4.45+gitAUTOINC+d00cdd80ab-r0 do_patch: Execution of '/home/pankaj/Oe-Raspberrypi4/tmp-glibc/work/raspberrypi4_64-oe-linux/linux-raspberrypi/1_5.4.45+gitAUTOINC+d00cdd80ab-r0/temp/run.do_patch.24212' failed with exit code 1:
(1/4) 0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch
[INFO]: check of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not pass, trying reduced context.
[INFO]: Context reduced git-am of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not work, trying "apply".
error: patch failed: tools/perf/util/srcline.c:191
error: tools/perf/util/srcline.c: patch does not apply
[ERROR]: Application of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch failed.
         Patch needs to be refreshed. Sample resolution script:
             .git/rebase-apply/resolve_rejects
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/pankaj/Oe-Raspberrypi4/tmp-glibc/work/raspberrypi4_64-oe-linux/linux-raspberrypi/1_5.4.45+gitAUTOINC+d00cdd80ab-r0/temp/log.do_patch.24212
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/pankaj/openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/pankaj/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/home/pankaj/openembedded-core/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['patch-native', 'quilt-native', 'kern-tools-native', 'autoconf-native', 'libtool-native', 'attr-native', 'automake-native', 'gnu-config-native', 'm4-native', 'texinfo-dummy-native', 'gettext-minimal-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_patch
| (1/4) 0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch
| [INFO]: check of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not work, trying "apply".
| error: patch failed: tools/perf/util/srcline.c:191
| error: tools/perf/util/srcline.c: patch does not apply
| [ERROR]: Application of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR: Could not apply patches for raspberrypi4-64.
| ERROR: Patch failures can be resolved in the linux source directory /home/pankaj/Oe-Raspberrypi4/tmp-glibc/work-shared/raspberrypi4-64/kernel-source)
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/pankaj/Oe-Raspberrypi4/tmp-glibc/work/raspberrypi4_64-oe-linux/linux-raspberrypi/1_5.4.45+gitAUTOINC+d00cdd80ab-r0/temp/run.do_patch.24212' failed with exit code 1:
| (1/4) 0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch
| [INFO]: check of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch with "git am" did not work, trying "apply".
| error: patch failed: tools/perf/util/srcline.c:191
| error: tools/perf/util/srcline.c: patch does not apply
| [ERROR]: Application of .kernel-meta//patches//./0001-perf-Make-perf-able-to-build-with-latest-libbfd.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/pankaj/meta-raspberry/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6107 tasks of which 6098 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/pankaj/meta-raspberry/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_patch
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.

am i missing anything???can anyone help me on this??

kraj, 2020-06-18:
I have a 5.4.47 cooking, will post for review today

agherzan, 2020-06-21:
Should be fixed by https://github.com/agherzan/meta-raspberrypi/pull/659.

closed 2020-06-21 by agherzan