agherzan/meta-raspberrypi #667: Is there a way to patch the cmdline.txt

Valks, 2020-06-26:
I'm trying to modify cmdline.txt during the build phase but I'm struggling to find where cmdline.txt is generated. Can anyone point me in the right direction?

I'm mostly looking to change the console output to a different port so it doesn't show up during boot, i.e.

```dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait   logo.nologo ```
to
```console=ttyAMA1 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait   logo.nologo ```

agherzan, 2020-06-26:
It gets generated here: https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi.inc#L139
You could overwrite the default value defined in the same recipe https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi.inc#L30 by using a bbappend, local.conf etc.

closed 2020-06-26 by agherzan

Valks, 2020-06-26:
@agherzan Thanks! That's exactly what I was trying to find.