agherzan/meta-raspberrypi #669: Yocto image with uboot not working for revolution pi?

amod-amatya, 2020-07-03:
I have built a rpi-basic-image provided by the meta-raspberrypi layer for raspberrypi and revolutionpi connect+. Now for my implementation, I need to use uboot. With meta-raspberrypi this is also quite simple, just add 
`RPI_USE_U_BOOT = "1"` 
in local.conf. 
I have also set the machine variable as:
MACHINE = `"raspberrypi-cm3"`
With this configuration, the image successfully boots for raspberrypi but does not boot for revolution-pi connect+. Is it that revolution-pi does not have support for uboot? I have also posted the issue in the kunbus revolution pi forum but have not received any response. Any suggestion to make it work?

agherzan, 2020-07-03:
I haven't heard anyone testing that board so I wouldn't be able to give you feedback but I'm sure that if you ask the uboot guys they will know much more.

closed 2020-07-03 by agherzan

amod-amatya, 2020-07-09:
@agherzan I think it is not actually about the board, it's about the emmc. It probably does not work for raspberrypi when using emmc instead of sd card. I have seen a similar problem here [https://www.raspberrypi.org/forums/viewtopic.php?t=217897](url) without any solution as well.