agherzan/meta-raspberrypi #670: No access to the camera on the Pi4 b

SCVready, 2020-07-04:
**Description**
Creating a distro for the Raspberry Pi 4 B using yocto dunfell, I'm getting an error message when I'm trying to launch:
`
raspivid -d
`(error copied below)
and when I launch:
`vcgencmd get_camera`I get: `supported=0 detected=0`

But using Raspbian it works correctly.

**Steps to reproduce the issue:**
1. Clone a clean yocto project, using dunfell:
```
git clone -b dunfell git://git.yoctoproject.org/poky.git
git clone -b dunfell git://git.yoctoproject.org/meta-raspberrypi
git clone -b dunfell git://git.openembedded.org/meta-openembedded
```
2. Launch enviroment variables script: 

```
source poky/oe-init-build-env
```
3. Modify config files:

in local.conf add:
```
MACHINE = "raspberrypi4"
ENABLE_UART = "1"
VIDEO_CAMERA = "1"
IMAGE_INSTALL_append = " userland"
```
in bblayers.conf add:
```
meta-openembedded/meta-oe
meta-openembedded/meta-multimedia
meta-openembedded/meta-networking
meta-openembedded/meta-python
meta-raspberrypi
```
4. Little fix on packagegroup-rpi-test.bb:

```
wireless-regdb \ ----> wireless-regdb-static \
```
5. Bitbake:

```
bitbake rpi-test-image
```
**Describe the results you received:**
Launching: `raspivid -d` 
I get:
```
mmal: Cannot read camera info, keeping the defaults for OV5647                  
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:E)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component                                         
mmal: main: Failed to create camera component                                   
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and d
```
Launching: `vcgencmd get_camera`
I get: `supported=0 detected=0`
**Describe the results you expected:**
Launching: `raspivid -d` 
Get no errors.
Launching: `vcgencmd get_camera`
Get: `supported=1 detected=1`

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
It works with raspbian: 2020-05-27-raspios-buster-lite-armhf.img

NobuoTsukamoto, 2020-07-05:
I have the same problem.

Add local.conf and bitbake core-image-weston
```
VIDEO_CAMERA = "1"
KERNEL_MODULE_AUTOLOAD += "bcm2835-v4l2"
GPU_MEM = "512"
```

- Occurs on the `dunfell` and `zeus` branches.
- Same 32bit and 64bit
- Raspberry Pi 4 Model B (I have not checked it with Raspberry Pi 3/2/1).

I get `dmsg | grep bcm2835` 
```
[    3.024650] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    3.201590] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    3.201613] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    3.251221] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    3.263214] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    3.269040] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    3.269069] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    3.271035] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    3.271066] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    3.274723] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    3.274747] bcm2835-codec bcm2835-codec: Loaded V4L2 isp

```

NobuoTsukamoto, 2020-07-11:
@SCVready 
I was able to solve this problem.
Changed /boot/config.txt (Near line 460 to 480).

Before:
```
## start_x
##     Set to "1" to enable the camera module.
##
##     Enabling the camera requires gpu_mem option to be specified with a value
##     of at least 128.
##
##     Default 0
##
#start_x=0
```
After:
```
## start_x
##     Set to "1" to enable the camera module.
##
##     Enabling the camera requires gpu_mem option to be specified with a value
##     of at least 128.
##
##     Default 0
##
start_x=1
```

Probably "start_x=1" near the last line of config.txt is not read.

SCVready, 2020-07-12:
Thank you so much @NobuoTsukamoto, it is also working for me.

PR #678 (2021-01-12): rpi-config: Relocated "start_x=1" in config.txt
@NobuoTsukamoto figured out that It seems like "start_x=1" needs to be earlier in the file to work properly on the Rapberry Pi 4 B.

Fixes: #670
commit 74deec51 (2021-01-12): rpi-config: Relocated "start_x=1" in config.txt

It seems like "start_x=1" needs to be earlier in the file to work properly on the Rapberry Pi 4 B.

Fixes: #670
closed 2021-01-12 by agherzan