agherzan/meta-raspberrypi #675: ERROR: Nothing PROVIDES 'core-image-sato'

embdlover, 2020-07-06:
Hi,
i am trying to build another core-image-sato for RPI4 i have added all the required layers in bblayer and also i have added all necessary features and install packages, i am attaching the the installs that i have added in local.conf

EXTRA_IMAGE_FEATURES ?= " debug-tweaks "
ENABLE_UART = "1"
CORE_IMAGE_EXTRA_INSTALL += "git ruby bsdtar pseudo os-release lsb-release openssh apt vim make  sysstat numactl python3 "
IMAGE_FEATURES_append = "tools-debug tools-sdk dev-pkgs"
IMAGE_INSTALL_append = " glibc-utils localedef unicode-ucd "
DISTRO_FEATURES_append =  "opengl python3-pip"
DISTRO ?= "exaleap-arm-linux"
IMAGE_INSTALL_append  = " glibc-utils localedef unicode-ucd "
IMAGE_FEATURES_append  = "tools-debug tools-sdk dev-pkgs"
DISTRO_FEATURES_append +=  " x11 "
MACHINE_FEATURES_append += "vc4graphics"
IMAGE_INSTALL += "glibc-staticdev"
IMAGE_INSTALL_append = "perf dbus libnss-mdns minicom rsync dbus dstat hostapd dhcpcd nasm ncurses-libncurses ncurses-tools ncurses-terminfo-base"
EXTRA_IMAGE_FEATURES_append  +=  "src-pkgs package-management ssh-server-openssh"
IMAGE_INSTALL_append  = "glibc-utils localedef unicode-ucd "
XTRA_IMAGE_FEATURES ?=  "allow-empty-password allow-root-login bash-completion-pkgs dbg-pkgs debug-tweaks dev-pkgs doc doc-pkgs eclipse-debug empty-root-password hwcodecs nfs-client nfs-server package-management post-install-logging ptest-pkgs splash src-pkgs ssh-server-dropbear ssh-server-openssh staticdev-pkgs tools-debug tools-profile tools-sdk tools-testapps x11 x11-base x11-sato"

# EXTRA_IMAGE_FEATURES  += "staticdev-pkgs "
USER_CLASSES ?= "buildstats image-mklibs image-prelink"

but am getting following error,

ERROR: Nothing PROVIDES 'core-image-sato'
core-image-sato was skipped: 'dev-pkgstools-debug' in IMAGE_FEATURES is not a valid image feature. Valid features: allow-empty-password allow-root-login bash-completion-pkgs dbg-pkgs debug-tweaks dev-pkgs doc doc-pkgs eclipse-debug empty-root-password hwcodecs nfs-client nfs-server package-management post-install-logging ptest-pkgs read-only-rootfs splash src-pkgs ssh-server-dropbear ssh-server-openssh stateless-rootfs staticdev-pkgs tools-debug tools-profile tools-sdk tools-testapps x11 x11-base x11-sato

With one image i successfully built this but with this image i am facing an issue,why i am getting this issue can you help??

pbrkr, 2020-07-06:
On Mon, 6 Jul 2020 at 12:53, embdlover <notifications@github.com> wrote:
> Hi,
> i am trying to build another core-image-sato for RPI4 i have added all the required layers in bblayer and also i have added all necessary features and install packages, i am attaching the the [...]

_append needs a leading space.

> IMAGE_INSTALL_append = " glibc-utils localedef unicode-ucd "
> DISTRO_FEATURES_append = "opengl python3-pip"

_append needs a leading space.

> DISTRO ?= "exaleap-arm-linux"
> IMAGE_INSTALL_append = " glibc-utils localedef unicode-ucd "
> IMAGE_FEATURES_append = "tools-debug tools-sdk dev-pkgs"

Looks like you've duplicated the above 2 lines.

> DISTRO_FEATURES_append += " x11 "
> MACHINE_FEATURES_append += "vc4graphics"

You're mixing _append and +=, you've got something confused here.

> IMAGE_INSTALL += "glibc-staticdev"
> IMAGE_INSTALL_append = "perf dbus libnss-mdns minicom rsync dbus dstat hostapd dhcpcd nasm ncurses-libncurses ncurses-tools ncurses-terminfo-base"

_append needs a leading space.

> EXTRA_IMAGE_FEATURES_append += "src-pkgs package-management ssh-server-openssh"

You're mixing _append and += again.

> IMAGE_INSTALL_append = "glibc-utils localedef unicode-ucd "

_append needs a leading space.

> XTRA_IMAGE_FEATURES ?= "allow-empty-password allow-root-login bash-completion-pkgs dbg-pkgs debug-tweaks dev-pkgs doc doc-pkgs eclipse-debug empty-root-password hwcodecs nfs-client nfs-server [...]

You're missing an E from the front of the variable name.
> EXTRA_IMAGE_FEATURES += "staticdev-pkgs "
>
> USER_CLASSES ?= "buildstats image-mklibs image-prelink"
>
> but am getting following error,
>
> ERROR: Nothing PROVIDES 'core-image-sato'
> core-image-sato was [...]

I recommend starting from scratch here. Your configuration is a bit of
a mess (happens to the best of us) and it's going to be very difficult
to reason about which statements are actually doing what.

You should be able to add just a few lines to local.conf to extend the
features and packages to be installed. If you're doing anything more
complex than that I usually recommend creating a distro config.

Thanks,

-- 
Paul Barker
Konsulko Group

embdlover, 2020-07-06:
EXTRA_IMAGE_FEATURES ?= "allow-empty-password allow-root-login bash-completion-pkgs dbg-pkgs debug-tweaks dev-pkgs doc doc-pkgs eclipse-debug empty-root-password hwcodecs nfs-client nfs-server package-management post-install-logging ptest-pkgs splash src-pkgs ssh-server-dropbear ssh-server-openssh staticdev-pkgs tools-debug tools-profile tools-sdk tools-testapps x11 x11-base x11-sato"

i  missed that E while copying here

Ok .thanks Paul for response, i will try to do on fresh build..

Thanks

embdlover, 2020-07-07:
Hi Paul,
I have made modifications,i have built the fresh raspberrypi4 core-image-sato  now i am not getting any error but my image is not booting, i have tried to check on the screen by connecting HDMI there i am able to see just 3 raspberry pi symbols after that i am not able to get anything on it

i have made following changes in local.conf

XTRA_IMAGE_FEATURES ?= "debug-tweaks"
ENABLE_UART  =  "1"
CORE_IMAGE_EXTRA_INSTALL += "git ruby bsdtar pseudo os-release lsb-release cmake vim sysstat python3 apt openssh"
DISTRO_FEATURES_append  = " python3-pip perf opengl "
# DISTRO_FEATURES_append  = " x11 "
IMAGE_INSTALL += "glibc-staticdev"
EXTRA_IMAGE_FEATURES_append  = " src-pkgs package-management ssh-server-openssh "
DISTRO ?= "exaleap-arm-linux"
IMAGE_INSTALL_append  = " glibc-utils localedef unicode-ucd "
EXTRA_IMAGE_FEATURES += "staticdev-pkgs "

I am missing anything which is needed to boot the image???

pbrkr, 2020-07-18:
You still have a mix of `+=` and `_append` statements. And some of those `_append` lines are missing a leading space in the appended string. Please review your local.conf file carefully, read the Yocto Project documentation and use `bitbake -e core-image-sato` to see what the final values of the variables are.

closed 2021-10-30 by agherzan