agherzan/meta-raspberrypi #689: u-boot not working on emmc on rasperry pi compute module3?

amod-amatya, 2020-07-21:
I have built a rpi-basic-image provided by the meta-raspberrypi layer for **raspberrypi compute module 3** with 16gb onboard emmc. For my implementation I need to use uboot. With meta-raspberrypi this is also quite simple, just add RPI_USE_U_BOOT = "1"
in local.conf. But with this configuration, the image does not seem to boot and if I disable u-boot, it is working fine.

PS. The same configuration works for raspberry pi 3 model b rev 1.2.

ithier, 2020-07-21:
What error are you getting ? 
Is U-Boot using the correct device tree ?

amod-amatya, 2020-07-21:
I am actually using a revpi connect+ which has raspberrypi compute module 3 and it does not have a serial console. I have a micro hdmi connected but it shows a blank screen.
The boot files are here:
[image]

The difference in the boot files when using with uboot vs without uboot

```
>> diff --brief --recursive  boot/ uboot/
Only in uboot/boot: boot.scr
Files boot/boot/extlinux/extlinux.conf and uboot/boot/extlinux/extlinux.conf differ
Files boot/boot/kernel7.img and uboot/boot/kernel7.img differ
Only in uboot/boot: uImage
```

The content of extlinux.conf with uboot enabled

`default Yocto
label Yocto
   kernel /uImage //without uboot here kernel is kernel7
   fdtdir /
append root=/dev/mmcblk0p2 rootwait 
` 

The content of boot.scr is [here](https://github.com/agherzan/meta-raspberrypi/blob/zeus/recipes-bsp/rpi-u-boot-scr/files/boot.cmd.in)

ithier, 2020-07-21:
I haven't successfully got U-Boot to work on the rpi 4 with this default boot.scr. I actually came here to ask how to fix it and found your question.
With the default boot.scr, I get a kernel panic, is that what you're getting ?

When I change my boot.scr to the following however, the kernel boots successfully (but I have a different problem that I am still trying to iron out, namely, the usb keyboard doesn't work):
```
setenv fdtfile bcm2711-rpi-4-b.dtb
mmc dev 0
fatload mmc 0:1 ${kernel_addr_r} Image
fatload mmc 0:1 ${fdt_addr_r} ${fdtfile}
setenv bootargs dwc_otg.lpm_enable=0 console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait noinitrd
booti ${kernel_addr_r} - ${fdt_addr_r}
```

What might be interesting to you, is that here I am explicitly loading the dtb for the pi4, maybe you need to do something similar but with: `bcm2710-rpi-cm3.dtb`

Hopefully someone more experienced can chime in and explain how that default boot.scr is chosing the correct device tree.

amod-amatya, 2020-07-21:
I tried explicitly loading bcm2710-rpi-cm3.dtb but still no help,
I am not sure whether its a kernel panic because i dont see anything except a blank screen.
my main concern: is it a problem with emmc,
I have found question similar to mine in rapsperrypi forum[https://www.raspberrypi.org/forums/viewtopic.php?t=217897](url)
but with no solution.

amod-amatya, 2020-07-21:
@ithier for me Its working now, but what I did was everything manually, I built an image without u-boot, and manually installed uboot and copied it into the boot dir, you can refer here for manual [installation](https://www.beyondlogic.org/compiling-u-boot-with-device-tree-support-for-the-raspberry-pi/), then I copied the uImage into the boot directory as well,
so, now when I started the pi, I got a uboot command line.
The bootscipt that you had posted I entered those commands manually and seems like it worked.
maybe I missed out somethng when integrating everything, I will double check it to make sure if it works automatically as well,
you can also try the manual way if that works for you..

ithier, 2020-07-21:
Nice!
Yes this works for me. Something must be wrong with the u-boot recipe in this layer ?

FYI you can automate the boot process by putting those commands in a file and compiling it into a boot.scr following [these instructions](https://elinux.org/RPi_U-Boot#boot.scr.uimg).
Then just replace the boot.scr script in the boot partition and it should run them automatically on boot.

amod-amatya, 2020-07-21:
Thank you @ithier , I will do that..

ithier, 2020-07-23:
I had time to play around a bit more with u-boot in this build and seem to have gotten it to work without modifying the boot.scr
**Can you retry using the original files from the build but make sure that you have `kernel=u-boot.bin` in your config.txt in the boot partition ?**
I noticed that the config being generated does not set this. Make sure it is set to `kernel=u-boot.bin`  and not `kernel="u-boot.bin"` too, it doesn't seem to work with the quotation marks.

Here is a summary of how I prepared my sd card for the rpi4 that worked (note: all files are found in the deploy directory):
* Prepare an SD card with the following commands (adjust for mmc device):
```
parted -s /dev/mmcblk0 mklabel msdos mkpart primary fat32 1M 100M mkpart primary ext4 100M 100%
mkfs.vfat /dev/mmcblk0p1
mkfs.ext4 /dev/mmcblk0p2
fatlabel /dev/mmcblk0p1 BOOT
e2label /dev/mmcblk0p2 ROOT
```
* Copy boot.scr into BOOT
* Copy u-boot.bin into BOOT (make sure to rename)
* Copy Image (or uImage) into BOOT (make sure to rename) 
* Copy the correct device tree into boot, in the case of the rpi4 it is `bcm2711-rpi-4-b.dtb`
* Copy the contents of the bcm2855-bootfiles directory into BOOT
* Extract the rootfs into ROOT

ithier, 2020-07-23:
You might also need to add `console=tty0` or `console=tty1` to cmdline.txt since you have no serial.

amod-amatya, 2020-07-28:
@ithier I was on a leave, sorry for late reply. The actual problem is similar to [this](https://github.com/agherzan/meta-raspberrypi/issues/528). I am on zeus branch which has the u-boot version 2019.07  which does not seem to work.
I switched to u-boot version 2018.09 and everything is working fine without changing anything.

amod-amatya, 2020-07-30:
@ithier Yes, like you said something is wrong with the u-boot here,
changing to a older version of u-boot works fine. Currently I am using uboot 2018.11 version.

agherzan, 2020-07-30:
@amod-amatya Are you able to run a bisect on uboot to figure out the offending commit?

amod-amatya, 2020-07-30:
@agherzan As I did not have access to serial console on revolution pi (rasberrypi cm3 with emmc) and connecting to monitor showed blank screen, i exactly could not tell where the problem was and it was working well with standard raspeberrypi 3b v1.2. Then, switching to a uboot version 2018.11 worked absolutely fine, so I am assuming uboot version to be a problem but I have not yet bisected on uboot. For now, I will stick with the older version.

amod-amatya, 2020-07-30:
@agherzan I think this issue can be closed now as it might also be a problem specific to the board

closed 2020-07-30 by amod-amatya

agherzan, 2020-07-30:
Sounds good.

Kottoz, 2021-04-14:
U-boot now supports rpi-4
```
kotoz@kotoz-Legion-Y540-15IRH:~/u-boot$ ls configs/rpi_
rpi_0_w_defconfig       rpi_3_32b_defconfig     rpi_3_defconfig         rpi_4_defconfig         rpi_defconfig           
rpi_2_defconfig         rpi_3_b_plus_defconfig  rpi_4_32b_defconfig     rpi_arm64_defconfig 
```

agherzan, 2021-04-14:
True, and that for a good while now.