agherzan/meta-raspberrypi #700: Feature request: raspi4 64 bit upstream 5.8.x+ kernel

RobertBerger, 2020-08-06:
** Description**

**Steps to reproduce the issue:**
* Using poky dunfell and meta-raspberrypi dunfell I hacked my way though meta-raspberrypi and it's possible to use the raspberry kernel 4.19, 5.4 and 5.8-rc6 and boot with a rootfs over nfs.

**Describe the results you received:**
* I added to upstream device tree overlays 
* I used upstream bcm2711-rpi-4-b.dts and it's dependencies
* Then I tried to use upstream 5.8 with defaultconfig, as well as bcm2711_defconfig - no luck

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

Is someone here interested to get a minimal rapberry 4 to run from an upstream kernel?

This is what I figured out in the meantime:
*) U-Boot has no network support, so Kernel and device tree need to reside on the SD card
*) rootfs over nfs works (with a happy raspi kernel)

Additional hints on how to get upstream to boot would be very useful.

RobertBerger, 2020-08-06:
Having said that I think there is lots of magic going on. What e.g. is kernel8.img?

mirzak, 2020-08-07:
`kernel8.img` is the file that the boot firmware will load when booting, e.g if you are using U-Boot in your build then `kernel8.img == u-boot.bin`.

The RPi boot firmware also supports booting the Linux kernel directly in which case `kernel8.img == Image`.

I know there is some strange management of the UART baudrate because it depends on [VPU frequencies](https://www.raspberrypi.org/documentation/configuration/uart.md), and it is normally setup by the boot firmware. E.g U-boot relies on that it is configured by the boot firmware, meaning it does attempt to re-configure it. A similar case could be with the Linux kernel. I am not sure if it is allowed to be changed UART configuration and there might be some special handling in the downstream kernel for this.

pbrkr, 2020-08-07:
You may want to look at how I tested upstream on raspberrypi3 in meta-kernel:

* Config to be included in local.conf or a distro conf file: https://gitlab.com/openembedded/community/meta-kernel/-/blob/master/conf/meta-kernel/bsp/raspberrypi3.inc
* Standalone recipe to populate cmdline.txt (I need to upstream this when I get some time): https://gitlab.com/openembedded/community/meta-kernel/-/blob/master/recipes-bsp/raspberrypi/rpi-cmdline.bb

RobertBerger, 2020-08-07:
> * Config to be included in local.conf or a distro conf file: https://gitlab.com/openembedded/community/meta-kernel/-/blob/master/conf/meta-kernel/bsp/raspberrypi3.inc

As far as I understand with a multi-v7 defconfig it is 32 bit and not 64 bit and it seems to be raspi 3 and not raspi 4.

Do you have upstream on raspi4 with 64 bit working?

pbrkr, 2020-08-07:
@RobertBerger I've not looked at rpi4 64-bit yet in meta-kernel. It should just be a case of finding the right defconfig if it is already supported upstream.

RobertBerger, 2020-08-07:
> @RobertBerger I've not looked at rpi4 64-bit yet in meta-kernel. It should just be a case of finding the right defconfig if it is already supported upstream.

It does not seem to be that easy. For arm 64 it's defconfig ;)

At the moment I copied a couple of patches (most likely more than actually needed) over to upstream from the raspi kernel and I got it to boot with patched upstream + defconfig but rootfs only from SD card - not yet over nfs.

RobertBerger, 2020-08-08:
Here is something I hacked up, which seems to work for me ;)

https://gitlab.com/meta-layers/meta-raspberrypi-ml-bsp/-/tree/dunfell/

closed 2020-08-08 by RobertBerger