agherzan/meta-raspberrypi #701: Question regarding bootfiles/overlays

LouDou, 2020-08-09:
**Description**

Hi,

I'm struggling to understand how to append additional overlay files into the image build for my application.
Specifically, I have a device with an Audioinjector Ultra and a Waveshare 5" HDMI touchscreen.

I have all the lines I need added to config.txt, but the required devicetree overlay files are not included by default in the build.
These overlay files are available in https://github.com/raspberrypi/firmware, but only a subset of these get included in the build.

They seem to be selected via the `RPI_KERNEL_DEVICETREE_OVERLAYS` variable.
I understand that perhaps not all files are included for size/space efficiency.

I'd like to extend this variable to add the files that I required. I cannot find any working examples or documentation for how to do this.

My application layer image config is here: https://bitbucket.org/doughammond/meta-rpi-guitar-pedal/src/master/
There's not a lot in there yet, as I'm still trying to get the basics correct before adding the actual apps on top.

Could you please advise?

Thanks,
Doug.

**Steps to reproduce the issue:**
1. build core-image-minimal for any rpi machine

**Describe the results you received:**

observe that not all overlays available in raspberrypi/firmware are not included in the final sd image files

**Describe the results you expected:**

Either:
- All overlay files include
OR
- Configuration available to select which overlays to include in the build

**Additional information you deem important (e.g. issue happens only occasionally):**

None.

**Additional details (revisions used, host distro, etc.):**

distro: poky
version: dunfell

agherzan, 2020-08-13:
Is the overlay currently built by all the kernel versions we have in the layer now? If yes, we could enable it by default. Otherwise you can just append it to `RPI_KERNEL_DEVICETREE_OVERLAYS` as part of your local.conf for example.

mterwoord, 2021-05-31:
I'm on thud.
I'm appending to `RPI_KERNEL_DEVICETREE_OVERLAYS` using:

```
RPI_KERNEL_DEVICETREE_OVERLAYS += " \
    overlays/uart1.dtbo \
	"
```

But now my overlays folder only contains the uart1 file, not the rest of the set.
To me it seems conf/machine/include/rpi-base.inc should set `RPI_KERNEL_DEVICETREE_OVERLAYS` using the `+=` operator to make appending (like you suggest) working?

kraj, 2021-05-31:
use `RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/uart1.dtbo"`

mterwoord, 2021-06-01:
Thanks. Not sure how I missed that.

closed 2021-06-08 by agherzan