agherzan/meta-raspberrypi #703: Raspberrypi4-64 WiFi is not working

matejsalamon, 2020-08-14:
**Description**

I'm trying to enable built-in wireless modul on RPi 4B+ using the latest meta-raspberrypi commit (bc7a06604d5232c57545bc7b2f57b9cce2699575) but I get always the same issue already during kernel boot. Should I change some settings using kernel menuconfig in order to do that or is it already ready to use for wlan connection? Can you suggest me please, how to get wireless modul running?

**Steps to reproduce the issue:**

[root@raspberrypi4-64 ~]$dmesg | grep brcmfmac
[    6.010129] brcmfmac: F1 signature read @0x18000000=0x15264345
[    6.018032] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.038176] usbcore: registered new interface driver brcmfmac
[    6.044087] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.bin failed with error -2
[    7.078037] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

[root@raspberrypi4-64 ~]$ifup wlan0
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
Could not read interface wlan0 flags: No such device
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
Cannot find device "wlan0"

**Describe the results you received:**
It seems that wireless kernel driver is not loaded at all.

**Describe the results you expected:**
Working wlan0 device on RPi4 64-bit using kernel and Yocto Linux

**Additional information you deem important (e.g. issue happens only occasionally):**
It happens always.

**Additional details (revisions used, host distro, etc.):**
poky (dunfell): 4b6faa9dada8d0ae7aba3a161429e4c1eab2fe21
meta-raspberrypi (dunfell): bc7a06604d5232c57545bc7b2f57b9cce2699575

kraj, 2020-08-14:
try it with tip of master

matejsalamon, 2020-08-15:
The same behaviour with tip of master.

agherzan, 2020-08-15:
Did this start to happen lately? I suspect a kernel update might have gone south. Are you up for a bisect?

texierp, 2020-08-15:
Wifi works fine on my side (RPi3):

```
root@raspberrypi3:~# ip -4 addr show wlan0
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel qlen 1000
    inet 192.168.1.51/24 brd 192.168.1.255 scope global wlan0
       valid_lft forever preferred_lft forever
```

So, what kind of image do you use with bitbake ? core-image-base ?

Maybe similar to #576

matejsalamon, 2020-08-16:
I tried to get wifi on a raspberrypi4-64 running for the first time. Before that I used always just wired eth port. Could it be that something goes wrong already with the boot-firmware due to the fact that I’m not using SD card any longer? My RPi boots and running Yocto built kernel and Linux on a SSD. 
I tried to use official rpi buster image and with that wifi works without any problems, but in this case kernel and Linux are 32-bit and they are on the SD card.
Sorry, I’m not native speaker therefore I didn’t understand your last question. Can you tell me what did you mean with it?

matejsalamon, 2020-08-16:
I’m using core-image-minimal. But I think the issue appears before Linux starts.

matejsalamon, 2020-08-16:
Hi,

Problem resolved! I replaced core-image-minimal by suggested core-image-base as you recommended. It is still not clear for me why does it work with that but anyway ...

Thank you very much

Matej

agherzan, 2020-08-18:
Because minimal (as far as I remember) doesn't include the fw files.

closed 2020-08-18 by agherzan