agherzan/meta-raspberrypi #706: no ethernet hardware found in current build

pabigot, 2020-08-23:
I've been away from Yocto for most of a year, and am trying to get an updated build with current master of poky, meta-openembedded, and meta-raspberrypi.

All went smoothly except the new system doesn't see the ethernet device at all.  This is reproducible on 3b, 3b+, and 4b, with both my image and core-image-minimal and rpi-test-image.

`ip link` shows lo and wlan0, but no eth0.

Booting with an old Yocto 2.5 for my image works fine, so it's probably not hardware or the network cable.

Reverting meta-raspberrypi to 5c3bda2acb8bd1c65e80ba9140e146f2f4a80b56 had no effect other than the kernel dropping from 5.4.58-v7 to 4.19.126-v7.

I'm sure it's something simple; can anybody provide a clue?  Thanks.

piechade, 2020-08-23:
In general it's working, there must be something deprecated or wrong used property somewhere in a layer or conf file.

kraj, 2020-08-24:
what does `ifconfig -a` show ?

pabigot, 2020-08-24:
```
root@raspberrypi3:~# ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:90 errors:0 dropped:0 overruns:0 frame:0
          TX packets:90 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7686 (7.5 KiB)  TX bytes:7686 (7.5 KiB)

wlan0     Link encap:Ethernet  HWaddr B8:27:EB:05:FE:0F  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@raspberrypi3:~#
```
A full capture of the boot and some commands including `lsmod` from `rpi-test-image` is available in https://gist.github.com/pabigot/35ad56a6a577af31dfc80f85055cbd76.

piechade, 2020-08-24:
@pabigot  try `sudo ip link set wlan0 up`

pabigot, 2020-08-24:
@piechade I'm not clear how that's going to help me get `eth0`, but here:

```
root@raspberrypi3:~# ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:90 errors:0 dropped:0 overruns:0 frame:0
          TX packets:90 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7686 (7.5 KiB)  TX bytes:7686 (7.5 KiB)

wlan0     Link encap:Ethernet  HWaddr B8:27:EB:05:FE:0F  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@raspberrypi3:~# rfkill unblock 0
root@raspberrypi3:~# [   40.834983] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
ip link set wlan0 up
root@raspberrypi3:~# ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:102 errors:0 dropped:0 overruns:0 frame:0
          TX packets:102 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8458 (8.2 KiB)  TX bytes:8458 (8.2 KiB)

wlan0     Link encap:Ethernet  HWaddr B8:27:EB:05:FE:0F  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@raspberrypi3:~#
```

piechade, 2020-08-24:
> @piechade I'm not clear how that's going to help me get `eth0`, but here:
>
> ```
> root@raspberrypi3:~# ifconfig -a
> lo        Link encap:Local Loopback
>           inet addr:127.0.0.1 [...]

I removed the answer already 😂, ignore it.

kraj, 2020-08-24:
> ```
> root@raspberrypi3:~# ifconfig -a
> lo        Link encap:Local Loopback
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING [...]

it seems to not find ethernet at all in kernel. This seems a bug to me. I will check here on rpi4

kraj, 2020-08-24:
@texierp it seems to be 5.4 issue have you encountered something like this on rpi3 ?

pabigot, 2020-08-24:
rpi4 seems to work (corrected original description).  I've only found this on rpi3.  Haven't tried rpi0.

Also I did go back to 4.19 and had the same problem, though I haven't re-verified that.

pabigot, 2020-08-26:
Looks like it was this:
```
# meta-raspberrypi special option to enable USB gadget mode
ENABLE_DWC2_PERIPHERAL = "1"
```
which was something I added eight months ago when I last touched yocto and was working only with rpi4.  Take that out and raspberrypi3 works.

closed 2020-08-26 by pabigot

kraj, 2020-08-26:
> Looks like it was this:
> 
> ```
> # meta-raspberrypi special option to enable USB gadget mode
> ENABLE_DWC2_PERIPHERAL = "1"
> ```
> 
> which was something I added eight months ago when I last touched yocto and was working only with rpi4. Take that out and raspberrypi3 works.

aw, that should have explained it on day 1, have you pointed out what you were holding back :) USB and ethernet are share controller on rpi3 so this would disable ethernet.

pabigot, 2020-08-26:
> > ENABLE_DWC2_PERIPHERAL = "1"
> 
> aw, that should have explained it on day 1, have you pointed out what you were holding back :) USB and ethernet are share controller on rpi3 so this would disable ethernet.

Yeah, I agree.  Problem is I thought I was starting from a working configuration.  Sorry about that.

agherzan, 2020-08-26:
That explains it.