agherzan/meta-raspberrypi #710: Backtrace on boot

piechade, 2020-08-28:
**Description**

I'm getting a Backtrace on boot and I'm not sure if it's related to poky/openembedded or meta-raspberrypi.

**Steps to reproduce the issue:**
1. Build raspberrypi image with lastest dunfell branch.
2. Boot the image on the pi

```[    1.511828] simple-framebuffer 1e513000.framebuffer: framebuffer at 0x1e513000, 0x6d8c00 bytes, mapped to 0x(ptrval)
[    1.511915] simple-framebuffer 1e513000.framebuffer: format=a8r8g8b8, mode=1824x984x32, linelength=7296
[    1.548182] Console: switching to colour frame buffer device 228x61
[    1.580552] simple-framebuffer 1e513000.framebuffer: fb0: simplefb registered!
[    1.581567] sysfs: cannot create duplicate filename '/devices/platform/1e513000.framebuffer'
[    1.581908] CPU: 0 PID: 1 Comm: swapper Not tainted 5.4.59 #1
[    1.582125] Hardware name: BCM2835
[    1.582269] Backtrace: 
[    1.582419] [<c00157dc>] (dump_backtrace) from [<c0015b28>] (show_stack+0x20/0x24)
[    1.582706]  r7:00000000 r6:cb5d87a0 r5:cb51e5a0 r4:cb47b000
[    1.582942] [<c0015b08>] (show_stack) from [<c07bc1b4>] (dump_stack+0x20/0x28)
[    1.583231] [<c07bc194>] (dump_stack) from [<c0252018>] (sysfs_warn_dup+0x60/0x74)
[    1.583519] [<c0251fb8>] (sysfs_warn_dup) from [<c025216c>] (sysfs_create_dir_ns+0xec/0x10c)
[    1.583820]  r7:00000000 r6:cb51e5a0 r5:ffffffef r4:cb5e2a10
[    1.584063] [<c0252080>] (sysfs_create_dir_ns) from [<c07c14e8>] (kobject_add_internal+0x90/0x308)
[    1.584385]  r6:c1039928 r5:c1039928 r4:cb5e2a10
[    1.584586] [<c07c1458>] (kobject_add_internal) from [<c07c1c98>] (kobject_add+0x60/0xc4)
[    1.584885]  r9:c0ac282c r8:c1078000 r7:00000000 r6:c1039928 r5:cb5e2a10 r4:00000000
[    1.585203] [<c07c1c38>] (kobject_add) from [<c0523a78>] (device_add+0x100/0x61c)
[    1.585474]  r3:00000000 r2:00000000
[    1.585738]  r6:c1039928 r5:dbf62b9c r4:cb5e2a10
[    1.585964] [<c0523978>] (device_add) from [<c0648c10>] (of_device_add+0x40/0x48)
[    1.586253]  r8:c1078000 r7:00000000 r6:00000000 r5:dbf62b9c r4:cb5e2a00
[    1.586514] [<c0648bd0>] (of_device_add) from [<c0649398>] (of_platform_device_create_pdata+0xb4/0xfc)
[    1.586857] [<c06492e4>] (of_platform_device_create_pdata) from [<c0649400>] (of_platform_device_create+0x20/0x24)
[    1.587221]  r7:c09c31e0 r6:c111e8f0 r5:00000000 r4:dbf62b9c
[    1.587462] [<c06493e0>] (of_platform_device_create) from [<c0aa366c>] (simplefb_init+0x74/0x9c)
[    1.587794] [<c0aa35f8>] (simplefb_init) from [<c000ade0>] (do_one_initcall+0x50/0x1d4)
[    1.588089]  r7:00000000 r6:c1078000 r5:c0ac284c r4:c0aa35f8
[    1.598984] [<c000ad90>] (do_one_initcall) from [<c0a731c0>] (kernel_init_freeable+0x130/0x1f4)
[    1.609929]  r7:c0a6f4a0 r6:c1078000 r5:c0ac284c r4:c0aeb94c
[    1.620971] [<c0a73090>] (kernel_init_freeable) from [<c07d3af0>] (kernel_init+0x18/0xfc)
[    1.631962]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c07d3ad8
[    1.642924]  r4:00000000
[    1.653955] [<c07d3ad8>] (kernel_init) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[    1.664988] Exception stack(0xcb4cbfb0 to 0xcb4cbff8)
[    1.675936] bfa0:                                     00000000 00000000 00000000 00000000
[    1.687241] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.698351] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.709317]  r5:c07d3ad8 r4:00000000
[    1.720413] kobject_add_internal failed for 1e513000.framebuffer with -EEXIST, don't try to register things with the same name in the same directory.
```

kraj, 2020-08-28:
Can you post `Build Configuration` from bitbake ? secondly is it using dunfell branch for all repos ?

piechade, 2020-08-28:
> Can you post `Build Configuration` from bitbake ? secondly is it using dunfell branch for all repos ?

Im using Dunfell for all branches. What exactly do you mean with build configuration? I'm using a custom image with own layer. There is nearly nothing in my local.conf

kraj, 2020-08-28:
> > Can you post `Build Configuration` from bitbake ? secondly is it using dunfell branch for all repos ?
> 
> Im using Dunfell for all branches. What exactly do you mean with build configuration? I'm using a custom image with own layer. There is nearly nothing in my local.conf

since the description does not have complete info to reproduce it, what bitbake prints at the beginning of build is a good information to stick in such reports always .

piechade, 2020-08-28:
> since the description does not have complete info to reproduce it, what bitbake prints at the beginning of build is a good information to stick in such reports always .

I edited the first post.

piechade, 2020-08-31:
@kraj do you have any idea to fix it. I have been looking for a solution for hours.

kraj, 2020-08-31:
it boots fine here on rpi4/32bit as well as 64bit so I dont know how can I help

piechade, 2020-08-31:
> it boots fine here on rpi4/32bit as well as 64bit so I dont know how can I help

Have somebody test it with raspberry pi 1?
Do you have some experience in debugging issues like this?

agherzan, 2020-09-08:
Does Raspbian boot on your board? Can you compare the kernel versions? Are you using uboot? If yes, can you test without it?

piechade, 2020-09-08:
> Does Raspbian boot on your board? Can you compare the kernel versions? Are you using uboot? If yes, can you test without it?

It's building in this moment a version without U-Boot and any customization. I need U-Boot for sw-update but for testing purpose I removed it also. The kernel version is 5.4.59.

I will give you a update when it's finished.

piechade, 2020-09-17:
Something new about the issue?

agherzan, 2020-09-18:
Sadly there is nobody now working on RPi1. I roughly know what the issue is but I recommend you bring it up to the uboot guys as they will be able to fix it in upstream and we can just update and take advantage of the fix.

piechade, 2020-09-19:
> Sadly there is nobody now working on RPi1. I roughly know what the issue is but I recommend you bring it up to the uboot guys as they will be able to fix it in upstream and we can just update and take advantage of the fix.

What I get as an answer from u-boot IRC.

what does RPI_USE_U_BOOT = "1" relate to? I would have expected kernel=u-boot.bin in config.txt. Which version of U-Boot are you using? Which U-Boot commands do you use for booting? You can find the address of the RPi maintainer for U-Boot in file /board/raspberrypi/rpi/MAINTAINERS. Maybe you want to send a mail with all details to him and the U-Boot list.

 sysfs: cannot create duplicate filename '/devices/platform/1e513000.framebuffer'
 this is your problem
dump the DT passed to the kernel and check for duplicate nodes for that framebuffer device

piechade, 2020-09-25:
> Sadly there is nobody now working on RPi1. I roughly know what the issue is but I recommend you bring it up to the uboot guys as they will be able to fix it in upstream and we can just update and take advantage of the fix.

I talked a lot with the U-Boot guys and the sad everything is working. The recommended me to find the issue yourself. Thereupon I have also tried to fix it, but my knowledge Bout the Linux kernel is to bad. You have mentioned that you know about the issue, can you give me some hints? Which recipes I need to modify with bitbake. I looked in U-Boot and there is nothing about a framebuffer definition. It's also possible that the framebuffer is defined in the prebuild DT files from raspberry pi firmware.

agherzan, 2021-10-30:
@piechade any progress on this?

closed 2021-11-13 by agherzan