agherzan/meta-raspberrypi #711: Raspberry Pi 4: OpenGL broken on X11

julianscheel, 2020-08-28:
Using OpenGL on X11 is mostly broken for Raspberry Pi 4 using v3d driver on systems built with meta-raspberrypi layer.

**Steps to reproduce the issue:**
1. Build an image including `mesa-demos` and and `xorg-server` for `MACHINE=raspberrypi4`
2. Boot system on Raspberry Pi 4
3. Start X and glxgears: `export DISPLAY=:0; X &; glxgears;`

**Describe the results you received:**

The output of glxgears is garbled.

**Describe the results you expected:**

glxgears should render it's gears.

**Additional details (revisions used, host distro, etc.):**

This is caused by the patch added in #558. Removing that patch makes GL work as expected.

kraj, 2020-08-28:
I gather you are building with 'vc4graphics' am I right ?
how about when using wayland ?

julianscheel, 2020-08-31:
Yes, that's with vc4graphics. For userland mesa/gallium is not used.
I don't have wayland in my builds, would have to do one to check.

xyklex, 2020-09-15:
Wow @julianscheel, thanks for creating this issue.

This is happening when using `vc4graphics`, without it `mesa-demos` recipe is not registered. 

This issue was the responsible of this one I registered [here](https://gitlab.gnome.org/GNOME/mutter/-/issues/1421), thinking it was an issue with mutter.

I did removed the patch #558 mentioned and glxgears worked as expected, this was the same output I got when running mutter. I think #723 is no longer relevant as I just realized that `userland` is a closed source library and it will not integrate smoothly with other apps.

kraj, 2020-09-15:
@xyklex please send a pull to revert this patch which was added in #558 
@@balav08 please debug this issue and propose a fixed patch, we will remove it until then

xyklex, 2020-09-15:
@kraj PR #724

shr-project, 2020-10-22:
FWIW: meta-webosose has a patch for mesa which is applied on top of the patch from #558 
https://github.com/webosose/meta-webosose/commit/5e1b6b208c0832e51dce6f4b50bd0cd18b4218b8
but it's a bit messed-up as this depends on DRM_FORMAT_MOD_LINEAR modifier being passed to dri2_from_fds() from another webOS OSE specific patch from:
https://github.com/webosose/meta-webosose/commit/c3a90c11bd0c7bbe587f6d96f151c3638c9e1c86

agherzan, 2021-11-14:
Based on the discussion above this looks fixed/addressed. I'll close for now but feel free to comment if that is not the case.

closed 2021-11-14 by agherzan