agherzan/meta-raspberrypi #717: Raspberry Pi 4 - 64 bit: U-Boot networking status

RobertBerger, 2020-09-06:
**U-Boot does not seem to support networking on the Raspberry Pi 4 - 64 Bit**

**Steps to reproduce the issue:**
```
U-Boot> setenv autoload no
U-Boot> dhcp
No ethernet found.
U-Boot> ping 192.168.42.1
No ethernet found.
ping failed; host 192.168.42.1 is not alive
U-Boot> 
```

**Describe the results you received:**
```
No ethernet found.
```

**Describe the results you expected:**
```
pong
```

**Additional information you deem important (e.g. issue happens only occasionally):**
Happens always

**Additional details (revisions used, host distro, etc.):**
Are there any plans to fix this? 
If so what's the time frame?

RobertBerger, 2020-09-06:
Did somebody try this? 
https://patchwork.ozlabs.org/project/uboot/cover/20200117012047.31096-1-andre.przywara@arm.com/

agherzan, 2020-09-08:
I'd recommend you follow up with the development in the uboot community. Matthias would probably be the man to ask about this. Checking the patchset it looks pretty promising. Was that merged?

mbgg, 2020-09-08:
Yes that's merged. I'll need to push a few fixes, let me do that right now...

agherzan, 2020-09-08:
Thanks, @mbgg for the feedback. @RobertBerger can you give an updated uboot version a spin? Maybe one with the pending fixes too.

RobertBerger, 2020-09-08:
@agherzan I'll try to add 2020.07 from poky master-next and see how that goes. If I manage to compile and boot this version and things are still not happy with networking I guess @mbgg will be able to point me into the right direction.

RobertBerger, 2020-09-08:
I hacked in the 2020.07, but this seems still to miss a few things.

```
U-Boot 2020.07 (Jul 06 2020 - 19:22:53 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0
U-Boot>
U-Boot>
U-Boot>
U-Boot>
U-Boot> setenv autoload no
U-Boot> dhcp
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
U-Boot> reset
```

```
U-Boot 2020.07 (Jul 06 2020 - 19:22:53 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0
U-Boot> 
U-Boot> 
U-Boot> 
U-Boot> setenv serverip 192.168.42.1
U-Boot> setenv gatewayip 192.168.42.254
U-Boot> setenv ipaddr 192.168.42.76
U-Boot> ping 192.168.42.1
Using ethernet@7d580000 device

ARP Retry count exceeded; starting again
ping failed; host 192.168.42.1 is not alive
U-Boot> 
```

@mbgg do you have patches on offer?

mbgg, 2020-09-09:
Try to add this two:
https://patchwork.ozlabs.org/project/uboot/patch/20200717133200.136257-1-jason.wessel@windriver.com/
https://patchwork.ozlabs.org/project/uboot/patch/20200717133200.136257-2-jason.wessel@windriver.com/

RobertBerger, 2020-09-09:
I applied the two patches and it works! @mbgg and all others thanks a lot.
Now I can load kernel/fdt over tftp and use a rootfs over nfs - perfect.[1]
https://pastebin.com/GYxRFqqS
Please note, that the U-Boot line 
setenv serverip 192.168.42.108
is only needed because my dhcp server (pfsense) dishes out another IP for serverip, so I set it explicitly here. It's NOT a problem with U-Boot or your patches - it's my setup.
-------
Now @agherzan we can discuss how to integrate this in meta-raspberrypi.
I kind of use my own, layer with my own kernel recipe and u-boot in addition to meta-raspberrypi, but will post the updates to it.

dunfell (which is what I used):
You need to patch poky/meta/classes/cml1.bbclass, since the new u-boot recipe needs find_cfgs, which comes from there
You need the u-boot 2020.07 recipe 
You need the two u-boot patches 

(poky) master/master-next (I didn't try), but I think:
You need only the two u-boot patches, which could be added to meta-raspberrypi

As soon as I have my layer under git I'll add it here as a reference as well.

@agherzan if you want me to hack up something for meta-raspberrypi let me know what you would like.

Once again - Thank you and definitely community rocks - chip/board vendors rock significantly less. 

(This whole story also deserves some entry in my blog as an example well how the community works: less than 3 days from no network to network working).

RobertBerger, 2020-09-15:
As promised: https://yoctoproject.blogspot.com/2020/09/raspberrypi-4-meta-raspberrypidunfell.html

agherzan, 2020-10-21:
If the work is already merged in upstream we will inherit it as soon as yocto updates uboot. So we will get it for free "soon". Otherwise, we could backport the relevant patches until that happens (if somebody is interested in providing a PR).

agherzan, 2021-10-30:
I would expect that this is already in the upstream oe-core version. Closing for now.

closed 2021-10-30 by agherzan