agherzan/meta-raspberrypi #723: mutter build fails with userland

xyklex, 2020-09-15:
I researched a little bit and it looks like this is a header provided by `mesa` but don't know how it should be included into the `userland` recipe.

**Description**
I'm building an image with `MACHINE_FEATURES_remove="vc4graphics"` so it will use `userland` as the `PREFERRED_PROVIDER` for `virtual/libgles2` and `virtual/egl`

**Describe the results you received:**
```
| cogl/cogl/cogl-egl-defines.h:38:10: fatal error: EGL/eglmesaext.h: No such file or directory
|    38 | #include <EGL/eglmesaext.h>
|       |          ^~~~~~~~~~~~~~~~~~
| compilation terminated.
```

**Describe the results you expected:**
Successfully compilation

kraj, 2020-09-15:
Can you add a bbappend for cogl in meta-raspberrypi which has something like 
```
DEPENDS_append_rpi = " ${@bb.utils.contains('MACHINE_FEATURES', 'vc4graphics', '', 'userland', d)}"
```
does that help?

xyklex, 2020-09-16:
I tried it as is, but unfortunately it does not worked, I appending it on cogl recipe first but then realized that it should actually be in mutter (it does not depends from the cogl recipe but it appears to has their own version).

Do you think this should still works with `userland`?

kraj, 2020-09-16:
> I tried it as is, but unfortunately it does not worked, I appending it on cogl recipe first but then realized that it should actually be in mutter (it does not depends from the cogl recipe but it appears to has their own version).
> 
> Do you think this should still works with `userland`?

I think yes it should work, with this fix applied and perhaps maybe there is some other quirk needed but I think it should work.

xyklex, 2020-09-21:
@kraj maybe is worthy to reopen this issue, I did what you recommended but unfortunately is still not compiling.

this is the patch generating the issue https://gitlab.gnome.org/GNOME/mutter/-/issues/876

and this is the error after I removed the patch
```
arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot -Icogl/cogl/d9c41d2@@mutter-cogl-5@sha -Icogl/cogl -I../mutter-3.34.6/cogl/cogl -Icogl -I../mutter-3.34.6/cogl -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/glib-2.0 -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/lib/glib-2.0/include -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/interface/vmcs_host/linux -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/interface/vcos/pthreads -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/libdrm -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/cairo -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/pixman-1 -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/uuid -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/freetype2 -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/libpng16 -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/gdk-pixbuf-2.0 -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/libmount -I/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot/usr/include/blkid -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -O2 -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0=/usr/src/debug/mutter/3.34.6-r0 -fdebug-prefix-map=/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0=/usr/src/debug/mutter/3.34.6-r0 -fdebug-prefix-map=/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot= -fdebug-prefix-map=/home/ubuntu/Yocto/RaspberryPi/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mutter/3.34.6-r0/recipe-sysroot-native= -fPIC -pthread -DUSE_VCHIQ_ARM '-DCOGL_LOCALEDIR="/usr/share/locale"' -DCOGL_COMPILATION '-DCOGL_GL_LIBNAME="libGL.so.1"' '-DCOGL_GLES2_LIBNAME="libGLESv2.so"' -MD -MQ 'cogl/cogl/d9c41d2@@mutter-cogl-5@sha/winsys_cogl-winsys-egl-x11.c.o' -MF 'cogl/cogl/d9c41d2@@mutter-cogl-5@sha/winsys_cogl-winsys-egl-x11.c.o.d' -o 'cogl/cogl/d9c41d2@@mutter-cogl-5@sha/winsys_cogl-winsys-egl-x11.c.o' -c ../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c
../MUTTER-3.34.6/COgl/cogl/winsys/cogl-winsys-egl-x11.c: In function '_cogl_winsys_egl_get_display':
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:269:7: error: unknown type name 'PFNEGLGETPLATFORMDISPLAYEXTPROC'; did you mean 'PFNGLXGETCURRENTDISPLAYEXTPROC'?
  269 |       PFNEGLGETPLATFORMDISPLAYEXTPROC get_platform_display =
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       PFNGLXGETCURRENTDISPLAYEXTPROC
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:270:2: warning: initialization of 'int' from 'void *' makes integer from pointer without a cast [-Wint-conversion]
  270 |  (void *) eglGetProcAddress ("eglGetPlatformDisplay");
      |  ^
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:273:30: error: 'EGL_PLATFORM_X11_KHR' undeclared (first use in this function)
  273 |  dpy = get_platform_display (EGL_PLATFORM_X11_KHR, native, NULL);
      |                              ^~~~~~~~~~~~~~~~~~~~
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:273:30: note: each undeclared identifier is reported only once for each function it appears in
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:273:8: error: called object 'get_platform_display' is not a function or function pointer
  273 |  dpy = get_platform_display (EGL_PLATFORM_X11_KHR, native, NULL);
      |        ^~~~~~~~~~~~~~~~~~~~
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:269:39: note: declared here
  269 |       PFNEGLGETPLATFORMDISPLAYEXTPROC get_platform_display =
      |                                       ^~~~~~~~~~~~~~~~~~~~
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:281:7: error: unknown type name 'PFNEGLGETPLATFORMDISPLAYEXTPROC'; did you mean 'PFNGLXGETCURRENTDISPLAYEXTPROC'?
[... cut ...]
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:285:8: error: called object 'get_platform_display' is not a function or function pointer
  285 |  dpy = get_platform_display (EGL_PLATFORM_X11_KHR, native, NULL);
      |        ^~~~~~~~~~~~~~~~~~~~
../mutter-3.34.6/cogl/cogl/winsys/cogl-winsys-egl-x11.c:281:39: note: declared here
  281 |       PFNEGLGETPLATFORMDISPLAYEXTPROC get_platform_display =
```

and https://mail.gnome.org/archives/commits-list/2017-February/msg01403.html looks like the code not working using `userland`

xyklex, 2020-09-21:
could this also be the issue? https://github.com/raspberrypi/userland/issues/469

agherzan, 2021-11-14:
@xyklex this seems to be an old issue and I'm wondering if you managed to make any progress on it.

closed 2022-02-23 by agherzan