agherzan/meta-raspberrypi #725: vc4graphics support on morty build

moorthy-bs, 2020-09-16:
**is vc4graphics possible on morty builds?**

**Steps to reproduce the issue:**
1. MACHINE_FEATURES += "vc4graphics"
2.
3.

**Describe the results you received:**
Wayland client apps through a common error when tried to run on top of a wayland compositor
wl_drm@12: error 2: invalid name

**Describe the results you expected:**
Wayland client apps to be run on top of wayland compositor

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
DISTRO = rdk
Wayland compositor is "Westeros"

kraj, 2020-09-16:
morty might not be possible for vc4graphics

agherzan, 2020-09-16:
We don't work on or maintain morty anymore so I don't think many people can confirm/infirm this support. It's sadly one of those cases where patches are welcomed - I know it's not helpful but that's the situation.

moorthy-bs, 2020-11-02:
vc4graphics support works from mesa version > 17.x whereas morty based builds provide mesa 12.x

closed 2020-11-02 by moorthy-bs