agherzan/meta-raspberrypi #728: "sdk_depends_ext" fails

kamarya, 2020-09-19:
**Description**

SDK generation fails on **dunfell**

**Steps to reproduce the issue:**
1. Generate the SDK `bitbake <image_name> -c populate_sdk_ext`

**Describe the results you received:**
```
do_sdk_depends: The file /lib/firmware/brcm/brcmfmac43430-sdio.MUR1DX.txt is installed by both linux-firmware and linux-firmware-rpidistro, aborting
```
**Describe the results you expected:**
The SDK gets built as usual

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

anilrath51, 2022-08-02:
I think this issue should be closed
I have followed the below steps to reproduce this issue but unable to get the following error :
**do_sdk_depends**: The file /lib/firmware/brcm/brcmfmac43430-sdio.MUR1DX.txt is installed by both linux-firmware and linux-firmware-rpidistro, aborting

`$ git clone -b dunfell git://git.yoctoproject.org/poky.git
$ git clone -b dunfell https://github.com/agherzan/meta-raspberrypi.git
$ git clone -b dunfell https://git.openembedded.org/meta-openembedded$ git clone -b dunfell git://git.openembedded.org/openembedded-core
$ . ./oe-init-build-env
`
`/poky/build$ cat conf/local.conf

MACHINE ??= "raspberrypi"
IMAGE_FSTYPES += "tar.bz2 wic wic.gz wic.bmap "`

`/poky/build$ cat conf/bblayers.conf
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /poky/meta \
  /poky/meta-poky \
  /poky/meta-yocto-bsp \
  /poky/meta-raspberrypi \
  /poky/meta-oe \
  /poky/meta-openembedded \
  "`
`bitbake core-image-minimal -c populate_sdk_ext
Loading cache: 100% |#####################################################################################################################| Time: 0:00:02
Loaded 2354 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.18"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "dunfell:d695bd0d3dc66f2111a25c6922f617be2d991071"
meta-raspberrypi     = "dunfell:2081e1bb9a44025db7297bfd5d024977d42191ed"
meta-oe              = "dunfell:d695bd0d3dc66f2111a25c6922f617be2d991071"
meta-openembedded    = "dunfell:52cee67833d1975a5bd52e4556c4cd312425a017"

Initialising tasks: 100% |################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 1 Found 0 Missed 1 Current 1536 (0% match, 99% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 4165 tasks of which 4162 didn't need to be rerun and all succeeded.`

agherzan, 2022-08-02:
Thanks @anilrath51 for confirming. Closing.

closed 2022-08-02 by agherzan