agherzan/meta-raspberrypi #729: wlan0 interface not available in RPI3

bhargavthriler, 2020-09-23:
**Description**
Unable to get the 'wlan0' interface in RPI3

Build configs
```
Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1+snapshot-20200923"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:a69d74c842a0f55ffa1408fa058138ac9637a0c6"
meta-raspberrypi     = "master:0e9c32c8169d0c6f0486e4d2e1ac670b11382c9c"
meta-oe              = "master:e7bf8b6cb2eac640e1f71ef08eb9398e7df661d4"
meta-python2         = "master:2684086f91e7074324081196c8a5f9945d39650e"
meta-shell           = "master:5e3ff88f9b45684d0ad20fb52eadc5dad6d4b92c"
meta-networking      
meta-python          = "master:e7bf8b6cb2eac640e1f71ef08eb9398e7df661d4"
```

added following in local.conf following https://github.com/agherzan/meta-raspberrypi/issues/576
```
DISTRO_FEATURES_append = " bluez5 bluetooth wifi"
IMAGE_INSTALL_append = " linux-firmware-bcm43430 bluez5 i2c-tools python-smbus bridge-utils hostapd iptables wpa-supplicant"
```
Kernel info:
```
root@raspberrypi3-64:~# uname -a
Linux raspberrypi3-64 5.4.64-v8 #1 SMP PREEMPT Fri Sep 11 12:57:30 UTC 2020 aarch64 GNU/Linux
root@raspberrypi3-64:~# 
```

**Steps to reproduce the issue:**
Build RPI3 image using meta-raspberrypi layer

**Describe the results you received:**
```
root@raspberrypi3-64:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr B8:27:EB:EA:BE:47  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
root@raspberrypi3-64:~# ifconfig wlan0 up
ifconfig: SIOCGIFFLAGS: No such device
root@raspberrypi3-64:~# 
root@raspberrypi3-64:~# ls /sys/class/net/
eth0  lo
root@raspberrypi3-64:~# 
root@raspberrypi3-64:~# lsmod   
    Not tainted
ipv6 548864 18 [permanent], Live 0xffffffde82193000
```

**Describe the results you expected:**
wlan0 should be available

**Additional details (revisions used, host distro, etc.):**
Host : ubuntu 18.04

agherzan, 2020-10-05:
What image are you building?

bhargavthriler, 2020-10-05:
I am building a custom image which is extended from core-image-minimal. Here is my image recipe 
```
SUMMARY = "Custom image containing required packages"
DESCRIPTION = "Custom image containing required packages. created 17-09-2020"

# based on core image minimal
include recipes-core/images/core-image-minimal.bb

IMAGE_INSTALL += " \
    util-linux \
    openssh \
    myshell \
    libstdc++ \
    wpa-supplicant \
    linux-firmware-rpidistro-bcm43430 \
    iw \
"

IMAGE_FEATURES += "package-management"
PACKAGE_CLASSES ?= "rpm deb package_deb"
```

agherzan, 2020-10-05:
Can you try it from `core-image-base`?

bhargavthriler, 2020-10-05:
@agherzan I tried with `core-image-base` still wlan0 is not available

One more thing, I had `CONFIG_IKCONFIG=y`, `CONFIG_IKCONFIG_PROC=y` enabled in my layer I can see that `CONFIG_WLAN` enabled
```
root@raspberrypi3-64:~# zcat /proc/config.gz | grep -i wlan
CONFIG_WLAN=y
CONFIG_WLAN_VENDOR_ADMTEK=y
CONFIG_WLAN_VENDOR_ATH=y
CONFIG_WLAN_VENDOR_ATMEL=y
CONFIG_WLAN_VENDOR_BROADCOM=y
CONFIG_WLAN_VENDOR_CISCO=y
CONFIG_WLAN_VENDOR_INTEL=y
CONFIG_WLAN_VENDOR_INTERSIL=y
CONFIG_WLAN_VENDOR_MARVELL=y
CONFIG_WLAN_VENDOR_MEDIATEK=y
CONFIG_WLAN_VENDOR_RALINK=y
CONFIG_WLAN_VENDOR_REALTEK=y
CONFIG_WLAN_VENDOR_RSI=y
CONFIG_WLAN_VENDOR_ST=y
CONFIG_WLAN_VENDOR_TI=y
CONFIG_WLAN_VENDOR_ZYDAS=y
CONFIG_WLAN_VENDOR_QUANTENNA=y
CONFIG_USB_NET_RNDIS_WLAN=m
root@raspberrypi3-64:~# 
```

LOG
```
root@raspberrypi3-64:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr B8:27:EB:EA:BE:47  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@raspberrypi3-64:~# ls /sys/class/net/
eth0  lo
root@raspberrypi3-64:~# ifconfig wlan0 up
ifconfig: SIOCGIFFLAGS: No such device
root@raspberrypi3-64:~# lsmod
    Not tainted
```

Build LOG
```
$ bitbake core-image-minimal
Loading cache: 100% |########################################################################################################################################################################| Time: 0:00:00
Loaded 3960 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1+snapshot-20201005"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:a69d74c842a0f55ffa1408fa058138ac9637a0c6"
meta-raspberrypi     = "master:0e9c32c8169d0c6f0486e4d2e1ac670b11382c9c"
meta-oe              = "master:e7bf8b6cb2eac640e1f71ef08eb9398e7df661d4"
meta-python2         = "master:2684086f91e7074324081196c8a5f9945d39650e"
meta-shell           = "master:5e3ff88f9b45684d0ad20fb52eadc5dad6d4b92c"
meta-networking      
meta-python          = "master:e7bf8b6cb2eac640e1f71ef08eb9398e7df661d4"

Initialising tasks: 100% |###################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 3 Found 0 Missed 3 Current 1509 (0% match, 99% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 3981 tasks of which 3965 didn't need to be rerun and all succeeded.
```

bhargavthriler, 2020-10-13:
@agherzan 
After building with latest commits also I am not getting wlan0 interface.
Any solutions you can suggest ?

kraj, 2020-10-13:
what does `ifconfig -a ` say ?

bhargavthriler, 2020-10-13:
> what does `ifconfig -a ` say ?
```
root@raspberrypi3-64:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr B8:27:EB:EA:BE:47  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
root@raspberrypi3-64:~# uname -r
5.4.69-v8
root@raspberrypi3-64:~# ls /sys/class/net/
eth0  lo
```

agherzan, 2020-10-13:
Can you attach a dmesg boot log?

bhargavthriler, 2020-10-13:
> Can you attach a dmesg boot log?

```
U-Boot 2020.07 (Jul 06 2020 - 19:22:53 +0000)

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT... OK
In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@7e980000: USB DWC2
scanning bus usb@7e980000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
262 bytes read in 3 ms (85 KiB/s)
## Executing script at 02400000
17654272 bytes read in 736 ms (22.9 MiB/s)
## Flattened Device Tree blob at 2eff8d00
   Booting using the fdt blob at 0x2eff8d00
   Using Device Tree in place at 000000002eff8d00, end 000000002f002f69

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.69-v8 (oe-user@oe-host) (gcc version 10.2.0 (GCC)) #1 SMP PREEMPT Tue Oct 6 10:44:59 UTC 2020
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] percpu: Embedded 31 pages/cpu s88024 r8192 d30760 u126976
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 669900K/970752K available (9724K kernel code, 1048K rwdata, 3320K rodata, 3072K init, 954K bss, 38708K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 30859 entries in 121 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] random: get_random_bytes called from start_kernel+0x334/0x4c4 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000260] Console: colour dummy device 80x25
[    0.000332] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.000359] pid_max: default: 32768 minimum: 301
[    0.000567] LSM: Security Framework initializing
[    0.000843] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000939] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002434] Disabling memory control group subsystem
[    0.004526] ASID allocator initialised with 32768 entries
[    0.004812] rcu: Hierarchical SRCU implementation.
[    0.005533] EFI services will not be available.
[... 175 lines cut ...]
INIT: Entering runlevel: 5
Configuring network interfaces... [    4.939348] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
udhcpc: started, v1.32.0
udhcpc: sending discover
udhcpc: sending discover
udhcpc: sending discover
udhcpc: no lease, forking to background
done.
Starting system message bus: dbus.
Starting bluetooth: bluetoothd.
Starting HOSTAP Daemon: Configuration file: /etc/hostapd.conf
nl80211: 'nl80211' generic netlink not found
Failed to initialize driver 'nl80211'
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
Starting syslogd/klogd: done

Poky (Yocto Project Reference Distro) 3.1+snapshot raspberrypi3-64 /dev/ttyS0

raspberrypi3-64 login: root
root@raspberrypi3-64:~# 

```

twoerner, 2020-12-01:
Can you try again with an updated master (across all layers)? I had noticed an issue a while back where the firmware files weren't being added to a basic build and there my wlan0 wasn't found by default either. Looking through your bootlog.txt I see that the bcrmfmac things aren't loaded or being reported during your boot.

I worked around it by adding ${MACHINE_EXTRA_RRECOMMENDS} to the list of installed packages. I think in the mean time the issue has been fixed. I just built an image for rpi3 and wlan0 came up without the MACHINE_EXTRA_RRECOMMENDS trick.

bhargavthriler, 2020-12-03:
> Can you try again with an updated master (across all layers)? I had noticed an issue a while back where the firmware files weren't being added to a basic build and there my wlan0 wasn't found by [...]

Yes I just checked updated master across all the branches, wlan0 interface is available now. Somewhere something was broken.

closed 2020-12-03 by bhargavthriler