agherzan/meta-raspberrypi #730: Unable to build the dunfell branch with virtualization feature enabled.

mkilivan, 2020-09-24:
**Description**
Unable to build the dunfell branch with virtualization feature enabled.

**Steps to reproduce the issue:**

1. Checkout `dunfell` branch
2. local.conf 
```
   MACHINE ??= "raspberrypi3"
   DISTRO_FEATURES_append = " virtualization"
```
3. bblayers.conf
```
BBLAYERS ?= " \
  /home/build/poky/meta \
  /home/build/poky/meta-poky \
  /home/build/poky/meta-openembedded/meta-oe \
  /home/build/poky/meta-openembedded/meta-multimedia \
  /home/build/poky/meta-openembedded/meta-networking \
  /home/build/poky/meta-openembedded/meta-python \
  /home/build/poky/meta-openembedded/meta-filesystems \
  /home/build/poky/meta-virtualization \
  /home/build/poky/meta-raspberrypi \
  "
```
4. Build linux-raspberrypi recipe
```
$ source oe-init-build-env
$ bitbake linux-raspberrypi
```
**Describe the results you received:**
Build failed.

```
| ERROR. input file "cfg/virtio.scc" does not exist
| ERROR: could not process input files: /home/murat/git/chaos-build/build/poky/build-new/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_5.4.59+gitAUTOINC+423495b33e-r0/defconfig /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/powersave.cfg /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/xt-checksum.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/ebtables.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/vswitch.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/lxc.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/docker.scc /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/0001-Revert-selftests-bpf-Skip-perf-hw-events-test-if-the.patch /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/0002-Revert-selftests-bpf-Fix-perf_buffer-test-on-systems.patch cfg/virtio.scc
|        See /tmp/tmp.MIQRq7HJ1L for details
| ERROR: Could not generate configuration queue for raspberrypi3.
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/murat/git/chaos-build/build/poky/build-new/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_5.4.59+gitAUTOINC+423495b33e-r0/temp/run.do_kernel_metadata.5618' failed with exit code 1:
| ERROR. input file "cfg/virtio.scc" does not exist
| ERROR: could not process input files: /home/murat/git/chaos-build/build/poky/build-new/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_5.4.59+gitAUTOINC+423495b33e-r0/defconfig /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/powersave.cfg /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/xt-checksum.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/ebtables.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/vswitch.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/lxc.scc /home/murat/git/chaos-build/build/poky/meta-virtualization/recipes-kernel/linux/linux-yocto/docker.scc /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/0001-Revert-selftests-bpf-Skip-perf-hw-events-test-if-the.patch /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/files/0002-Revert-selftests-bpf-Fix-perf_buffer-test-on-systems.patch cfg/virtio.scc
|        See /tmp/tmp.MIQRq7HJ1L for details
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_kernel_metadata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 523 tasks of which 521 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/murat/git/chaos-build/build/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_kernel_metadata
```

**Describe the results you expected:**
Should build successfully.

**Additional information you deem important (e.g. issue happens only occasionally):**
The issue happens every time.

**Additional details (revisions used, host distro, etc.):**
Host : ubuntu 18.04

mkilivan, 2020-10-05:
Fixed by #731

closed 2020-10-05 by mkilivan

shr-project, 2020-10-06:
Do you want to have virtio enabled on rpi?

When I've noticed this in webOS I've just removed virtio.cfg in:
https://github.com/webosose/meta-webosose/commit/a8d9dff92610ad6e4fd8e40d6d5c89b15542c7cb#diff-6ccf8bc05c2664ecd3a7a651c585b947R12

and when I've asked Bruce about it in IRC, he said that it should be only an warning when KERNEL_DANGLING_FEATURES_WARN_ONLY is enabled:
> 23:42 < JaMa> zeddii: talking about meta-virt compatibility, is every linux-yocto kernel supposed to have cfg/virtio.scc in kernel-meta?
> 23:43 < JaMa> zeddii: it's included in http://git.yoctoproject.org/cgit/cgit.cgi/meta-virtualization/commit/?id=21d8bcdb791a1ea766a3e7e7663c7b6d49bc861a but seems like new 5.4 linux-raspberrypi kernel doesn't have this and fails because of it
> 04:01 < zeddii> JaMa: yes. that's the design. We were talking about this on the list the other day. And it's why I added the ability to make those dangling features a warning.
> 04:02 < zeddii> (KERNEL_DANGLING_FEATURES_WARN_ONLY)
> 04:24 < zeddii> that of course doesn't mean things can't be changed, but the error being thrown is known

And from the #yocto on freenode on 2020-09-04 it looks like linux-imx has similar issue, maybe it should be improved somehow in meta-virtualization instead? @zeddii ?

zeddii, 2020-10-06:
> Do you want to have virtio enabled on rpi?
>
> When I've noticed this in webOS I've just removed virtio.cfg in:
> webosose/meta-webosose@a8d9dff#diff-6ccf8bc05c2664ecd3a7a651c585b947R12
>
> and when I've asked Bruce about it in IRC, he said that it should be only
> an warning when KERNEL_DANGLING_FEATURES_WARN_ONLY is enabled:
>
> 23:42 < JaMa> zeddii: talking about meta-virt compatibility, is every
> linux-yocto kernel supposed to have cfg/virtio.scc in kernel-meta?
> 23:43 < JaMa> zeddii: it's included in
> http://git.yoctoproject.org/cgit/cgit.cgi/meta-virtualization/commit/?id=21d8bcdb791a1ea766a3e7e7663c7b6d49bc861a
> but seems like new 5.4 linux-raspberrypi kernel doesn't have this and fails
> because of it
> 04:01 < zeddii> JaMa: yes. that's the design. We were talking about this
> on the list the other day. And it's why I added the ability to make those
> dangling features a warning.
> 04:02 < zeddii> (KERNEL_DANGLING_FEATURES_WARN_ONLY)
> 04:24 < zeddii> that of course doesn't mean things can't be changed, but
> the error being thrown is known
>
> And from the #yocto on freenode on 2020-09-04 it looks like linux-imx has
> similar issue, maybe it should be improved somehow in meta-virtualization
> instead? @zeddii <https://github.com/zeddii> ?
Clearly the default for this is causing meta-virt users some pain, and
that's not what I want.

We are caught between asking for a generic kernel bbappend, so the
fragments can be used, but it is also picking up kernel's that don't use
the fragments.

One option is that I'll switch the default to a warning only, versus the
current error.

I'll also see if there's a better way that I can detect if the main kernel
meta data is being used, and only trigger the warning/error in that
scenario.

Bottom line: I'll fix this so meta-rpi doesn't have to adjust (if it
doesn't want to).

Bruce

PR #731 (2020-10-05): linux-raspberrypi_5.4.bb: Add kernel-cache source
Fixes the following error:
ERROR. input file "cfg/virtio.scc" does not exist
fixes #730 

The `linux-raspberrypi_5.4` recipe successfully builds after this change. This patch has been tested on the dunfell branch. Probably need to apply to the other branches once it's verified.

**- What I did**

**- How I did it**