agherzan/meta-raspberrypi #732: Rpi 4 doesn't boot with u-boot when enable_uart=0 in config.txt

tolszak, 2020-09-28:
Regular dunfell branch with local.conf

```
MACHINE = "raspberrypi4-64"
RPI_USE_U_BOOT=1
ENABLE_UART=0
```
core-image-minimal doesn't boot. It looks like uboot image hangs for some reason. Sometimes rainbow splash screen blinks.

I tried updated u-boot from July 2020 - no luck.
I removed serial from rpi.h ENV_DEVICE_SETTINGS in u-boot - no luck.

Spark1114, 2020-11-04:
I have kind of the same issue on the CM3+ with the following config: 

```
MACHINE = "raspberrypi3-64"
ENABLE_UART = "1"
RPI_USE_U_BOOT = "1"
```

And these layers: 

```
poky/meta-openembedded/meta-python
poky/meta-openembedded/meta-networking
poky/meta-openembedded/meta-oe
poky/meta
poky/meta-poky
poky/meta-yocto-bsp
poky/meta-raspberrypi
```

Here is the console output: 
                                        
```
U-Boot 2020.01 (Jan 06 2020 - 20:56:31 +0000)
                                        
DRAM:  948 MiB                          
RPI Compute Module 3+ (0xa02100)
```

I am able to boot the system when using MACHINE = "raspberrypi-cm3" 

Let me know if you need more information.

cveilleux, 2021-01-12:
Confirming we are seeing the same issue on raspberrypi4-64 with `RPI_USE_U_BOOT = "1"`

We have not set ENABLE_UART. I believe it is related to RPI_USE_U_BOOT that causes the boot failure.

HDMI screen plugged. Rainbow flashes quickly, then screen remains black.

agherzan, 2021-01-12:
Duplicates https://github.com/agherzan/meta-raspberrypi/pull/733

EasyIP2023, 2021-09-15:
I'm also having u-boot issues on the pi 4. Serial console displays something similar to this for me. Looks like u-boot support for the pi 4 may need to be updated. Not sure if the version of u-boot used in this layer is this https://github.com/agherzan/u-boot/tree/ag/rpi4 or upstream u-boot. Upstream u-boot doesn't seem to have too much support for the raspberrypi 4. @agherzan correct me if I'm wrong about above. Also it would be nice to have upstream u-boot support for the pi 4. Sure are there kernel changes and a number of extra steps to get u-boot working again on the pi 4.

```
DRAM:  3.9 GiB
RPI 4 Model B (0xc03112)
mbox: Timeout waiting for response
bcm2835: Could not set module 3 power state
initcall sequence 000000003b3d28f8 failed at call 0000000000084344 (err=-5)
### ERROR ### Please RESET the board ###
```

agherzan, 2021-09-17:
@EasyIP2023 You should not use that repo anymore. It was done for the initial support of rpi4 which was completely merged in upstream. That's why in this BSP layer we are resuing the uboot recipe from oe-core.

agherzan, 2022-02-23:
@tolszak is this still an issue on current versions?

agherzan, 2022-02-23:
Thanks @tolszak.

closed 2022-02-23 by agherzan