agherzan/meta-raspberrypi #734: Is it possible to encrypt the rootfs partition in the SD card image ?

myfreescalewebpage, 2020-09-29:
**Description**

Is it possible to encrypt the rootfs partition in the SD card image ? The goal is to prevent someone to retrieve the content of the SD card on another machine.

I have read several things about this subject (https://linux.die.net/man/8/cryptsetup for example) but the purpose is to do it with Yocto of course :)

**Describe the results you expected:**

rootfs partition encrypted

**Additional details (revisions used, host distro, etc.):**

Currently using dunfell but it's not a strong requirment

agherzan, 2020-10-05:
Sure it is. You can use an initramfs to unlock a luks device. Nevertheless, this solution is not RPi specific at all.

closed 2020-10-05 by agherzan

myfreescalewebpage, 2020-10-06:
@agherzan thanks for the answer, looks good if it is not rpi specific. Can you anyway point us a link/tuto if you know to help starting on this subject ? I haven't found pertinent content at this time...

agherzan, 2020-10-21:
Here it is: https://wiki.archlinux.org/index.php/dm-crypt