agherzan/meta-raspberrypi #737: Question about DEFAULTTUNE for raspberrypi 3/4 (32bit)?

dev-0x7C6, 2020-10-03:
Hi,

I'm playing for some time with raspberrypi3/4 and I found that 32-bit versions using `cortexa7` tune which may limit instruction set capabilities. Can I somehow extend those flags to pass proper cortex version?

From what I understand using `tune-cortexa57` or `tune-cortexa72` will change architecture to aarch64, so I should avoid them. Am I correct ?

Cheers,
Bartłomiej

kraj, 2020-10-03:
I think we use cortexa53 for rpi3 machines which translates to armv7ve architecture for 32bit so I am not sure how you are ending up with armv7a

dev-0x7C6, 2020-10-03:
This is a part of [raspberrypi3.conf](https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/raspberrypi3.conf)

```
#@TYPE: Machine
#@NAME: RaspberryPi 3 Development Board (32bit)
#@DESCRIPTION: Machine configuration for the RaspberryPi 3 in 32 bits mode

DEFAULTTUNE ?= "cortexa7thf-neon-vfpv4"
require conf/machine/include/tune-cortexa7.inc
include conf/machine/include/rpi-base.inc
```

@kraj It looks like `cortexa7` configuration for me. 

This is my bitbake output after using `raspberrypi3` as machine:
```
Build Configuration:
BB_VERSION           = "1.47.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "retro"
DISTRO_VERSION       = "3.1+snapshot-20201003"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta-dev1990         = "master:7c3718aad3415012905d8561f20b32379f06aa84"
meta-intel           = "master:b077acb2702731db2d77ed31126a5b21af3c1cff"
meta-kodi            = "master-next:8122bf7aaa11e40bd9c0048cdc0e21b7a24b7391"
meta-mesa            = "master:1e4126a1073b859092dc798b531a661f9c9ce5a8"
meta-multimedia      
meta-networking      
meta-oe              
meta-python          = "master:a599b850dc124444d06fbb5d48734c134325ac0c"
meta-python2         = "master:7b18fa3ead2c1a98e0f4e01ab57027f57d362c04"
meta-qt5             = "5.15:0108285c3fb4f246fc73e33afd6a34ba1eb76356"
meta-raspberrypi     = "master:0e9c32c8169d0c6f0486e4d2e1ac670b11382c9c"
meta-rauc            = "master:04ead4e4968a9fd97511d1bd105c372fc3b2a8c5"
meta-retro           = "master:62df7ffc2015c236ddf6ef8754ab6c266e5ff985"
meta                 
meta-poky            
meta-yocto-bsp       = "master:520e95902f12feac2850e0c24167002eb2cc7c42"
```

agherzan, 2020-10-05:
What tunes are you thinking of trying? I think we looked into this a long time ago and we realized that there was no real performance benefit.

dev-0x7C6, 2020-10-05:
> What tunes are you thinking of trying? I think we looked into this a long time ago and we realized that there was no real 
performance benefit.

`-mcpu=cortexa53` and `-mcpu=cortexa72` respectively  and `AArch32` (`Armv8-A`) architecture 

But lets assume for now that I'm asking this only from correctness stand point :-)

`raspberrypi3` has `cortexa7` tune, but it is a `cortexa53` processor, why not use just `cortexa53` ? (default to `AArch32` ?)
`raspberrypi4` has `cortexa7` tune, but it is a `cortexa72` processor, why not use just `cortexa72` ? (default to `AArch32` ?)

Things to consider:
* different compiler may generate a difference
* newer version of compiler may generate a difference
* combining flags may generate a difference
* tunes are created by compiler developers for a reason

Is there a reason why `cortexa7` is common for `raspberrypi3` and `raspberrypi4`? Shareable packages (why then 64-bit versions do not follow same route?)?

I'm currently researching if I can squeeze some performance with my current emulator library. Setting `-mcpu=cortex-a53` (`rpi3`) for one specific emulator did not improve performance (probably compiler targeting `armv7a` instruction set, which is not correct it should be `Armv8-A` `AArch32`, so I need to redone this). But I wonder if I add some vectorization flags it may work better with `-mcpu=cortex-a53` since this tells compiler what core is able to do. I think that `cortexa7` can only limit possible outcomes. But I probably should test both ;-)

dev-0x7C6, 2020-10-05:
I tested my local cross compiler to check what `-mcpu=cortex-a72` and `-mcpu=cortex-a53` is changing.

`armv7a-unknown-linux-gnueabihf-gcc -mcpu=cortex-a72 --help=target -Q`
result: `-march=armv8-a+crc+simd`

`armv7a-unknown-linux-gnueabihf-gcc -mcpu=cortex-a53 --help=target -Q`
result: `-march=armv8-a+crc+simd`

`armv7a-unknown-linux-gnueabihf-gcc -mcpu=cortex-a7 -mfpu=neon-vfpv4 --help=target -Q`
result:  `-march=armv7ve+simd`

in addition there is also possibility to set fpu to `crypto-neon-fp-armv8` to enable `crypto` instructions.

Some crypto libraries may use those extensions when compiler supports it in order to get processor acceleration. 

I searched for information if there is possibility to use more registers in 32-bit context (I searched for something similar to https://en.wikipedia.org/wiki/X32_ABI) but I think this is not possible here - probably not safe route. 

Hope this is enough to consider again `cortex-a53` and `cortex-a72` tunes.

kraj, 2020-10-05:
@agherzan it might make sense now though, especially for rpi4

agherzan, 2020-10-21:
I don't have the time to pick this up but I'll gladly welcome PRs.

dev-0x7C6, 2020-10-24:
Ok, I'll probably send patches in near future. Thanks.

closed 2020-10-24 by dev-0x7C6

agherzan, 2020-10-25:
Thanks @dev-0x7C6 . Good stuff.