agherzan/meta-raspberrypi #743: RPI3 build failing after commit "linux-raspberrypi_5.4.bb: Update to 5.4.69" "f762c8d9f5227f6d89180ead18af93d79ce8562b"

bhargavthriler, 2020-10-07:
**Description**
RPI3 builds are failing after  **linux-raspberrypi_5.4.bb: Update to 5.4.69** : `f762c8d9f5227f6d89180ead18af93d79ce8562b` commit.

After pulling the latest changes builds for RPI3 are failing with the following error.
`ERROR: Task (/home/bhargav/RPI3/Build/poky2/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_compile) failed with exit code '1'`

After checking individual commits I found out that this commit `f762c8d9f5227f6d89180ead18af93d79ce8562b` is causing the build failure

**NOTE**: Currently I have check only in RPI3-64 build. Building 32bit image now, will update this part once done.

**Steps to reproduce the issue:**
1. Clone poky from https://git.yoctoproject.org/git/poky
2. Clone meta-raspberrypi layer from https://github.com/agherzan/meta-raspberrypi.git
3. Add `MACHINE ??= "raspberrypi3-64"` in local.conf 
4. Build `core-image-base` or `core-image-minimal`

**Describe the results you received:**
Build failing 

Complete build log attached "build-failure-log.txt"
Error LOG
```
ERROR: Task (/home/bhargav/RPI3/Build/poky2/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_compile) failed with exit code '1'
```
**Describe the results you expected:**
Build should be successful.
[build-failure-log.txt](https://github.com/agherzan/meta-raspberrypi/files/5342620/build-failure-log.txt)

**Additional details (revisions used, host distro, etc.):**
Host : Ubuntu 18.04.4

kraj, 2020-10-07:
https://github.com/agherzan/meta-raspberrypi/pull/742 should fix it. Please test and report

bhargavthriler, 2020-10-07:
> #742 should fix it. Please test and report

Yes using this fix. Image is building successfully

agherzan, 2020-10-09:
Fixed in https://github.com/agherzan/meta-raspberrypi/pull/740

closed 2020-10-09 by agherzan